opulo-inc/lumenpnp #7: Add KiCad cache files to .gitignore

NightSkySK, 2020-05-12:
I'm not an expert about KiCad but I assume files like
pnp/pcb/mobo/fp-info-cache
pnp/pcb/mobo/*.sch-bak
pnp/pcb/mobo/*.kicad_pcb-bak
should be added to .gitignore and removed from repository?
Please refer to articles:
[Project and libary setup for sharing and collaboration](https://forum.kicad.info/t/project-and-libary-setup-for-sharing-and-collaboration/20842)
[fp-info-cache purpose](https://forum.kicad.info/t/fp-info-cache-purpose/16569)

Of course the same rules should apply to:
pnp/pcb/piggyback
feeder\pcb\feederFloor
feeder\pcb\indexingWheel
feeder\pcb\mobo

adlerweb, 2020-05-12:
Probably a good idea to just use common gitignore templates like https://www.gitignore.io/api/kicad

NightSkySK, 2020-05-12:
@adlerweb yes, it seems to be the best idea.  The only thing that @sphawes need to consider if he want to keep Gerbers available (which may ends like in issue #8 ) or like the templete suggest to exclude gereber folders 

``` 
# Gerber export output
out/
```

sphawes, 2020-05-15:
Thanks @NightSkySK @adlerweb, I'll make sure to add those to my gitignore. I'm inclined to stop exporting the Gerber files to prevent things like [#8](https://github.com/sphawes/index/issues/8) as you said. Makes them a bit less accessible, but given that it's still heavily under development and likely buggy, I don't think that's a bad thing.

sphawes, 2020-05-15:
Fixed with commit 701ddd6.

closed 2020-05-15 by sphawes

adlerweb, 2020-05-15:
> Makes them a bit less accessible, but given that it's still heavily under development and likely buggy, I don't think that's a bad thing.
For the meantime, yes. I would suggest to tag the repo and attach gerbers as "Pre-Release" if you produce a prototype for a video. This way people could access gerbers and create boards as shown and easily compare to current HEAD.