opulo-inc/lumenpnp #10: Open PNP Machine File?

bjonesrock, 2020-05-13:
Could you upload your machine file for Open PNP, and or provide some documentation for what settings you used?

Here are instructions on where to find your machine file:
[https://github.com/openpnp/openpnp/wiki/FAQ#where-are-configuration-and-log-files-located](url)

CesarMardueno, 2020-06-17:
I would like if someone has it to help us by sharing it the machine.xml file since in my case they already try in many ways and I do not get good results

raouldeheer, 2020-07-28:
@sphawes Could you please share the machine file?
So we can close this issue.

sphawes, 2020-08-02:
Attached is the machine file that i currently have loaded in OpenPnP, but _it is definitely not correct._ I still have a large body of tuning to do, and I believe some stuff in here is still assuming the Piggyback/RAMPS configuration, and not the full motherboard. I'm not adding to the repo because this is really unfinished, but if y'all want a starting point, this is what I currently have. I'll leave this open so I remember to add it to the repo once tuning is done.
[machine.xml.zip](https://github.com/sphawes/index/files/5012553/machine.xml.zip)

LoneWalkerWolf, 2020-11-01:
After spending over two hours on OpenPnP, I also attach my machine file along with a component example tooling file. Please note that this is still using the **RAMPS 1.4 configuration** and not the full developed motherboard _(as Stephen has said, it is unfinished)._ 
[openpnp-index.zip](https://github.com/sphawes/index/files/5470978/openpnp-index.zip)
This works really well for basic use, and has scripts for turning on and off the Ring Light on both cameras.

G-Pereira, 2021-01-30:
@JudgeBeeb Can you share your machine file as PR to the mobo-rev03 branch? You can put it in: `index/pnp/code/openpnp-config/machine.xml` (new folder)

wilhelmzeuschner, 2021-01-30:
I just created a repo to host my current config file. As with the others, this is still very much a work in progress and subject to changes. [index_pnp_config](https://github.com/wilhelmzeuschner/index_pnp_config)

JudgeBeeb, 2021-01-30:
I have uploaded mine too and created a PR.

G-Pereira, 2021-07-29:
I am working on a OpenPnP V2 machine configuration file as we speak!

G-Pereira, 2021-09-14:
#358 fixed this!

closed 2021-09-14 by G-Pereira