opulo-inc/lumenpnp #9: RAMPS Wiring Diagram?

bjonesrock, 2020-05-13:
I'm doing diff checks on Marlin to see what was changed, but it'd be super useful to know how you're wiring things up. Could you make a wiring diagram for RAMPS?

DurandA, 2020-05-13:
Not sure if @sphawes read this, but [one comment on Hackaday](https://hackaday.com/2020/05/11/open-source-pick-and-place-has-a-450-bom-cost/#div-comment-6244592) mentioned that RAMPS support is limited for OpenPNP. However, the officially supported Smoothieboard availability seems to be a bit limited.

bjonesrock, 2020-05-14:
I'm in the process of setting up my own PNP, the Smoothieboard seems to be the preferred controller for Open PNP but RAMPS is just so ubiquitous and affordable. 

Like you said it's hard to buy one right now, they just did a Kickstarter for Smoothieboard 2 and there isn't a release date yet. They also seem to be out of stock on Smoothieboard 1. 

I'm working through the modified Marlin firmware included in this project, but having a diagram of the default configuration would make it much easier to get the PNP wired up.

ncx94, 2020-05-14:
There is a custom MB in the works so I don't think there will be a ton of support for the Ramps :)

However it is still mega-based so I don't know how well it will work with OpenPNP.

@sphawes  Maybe consider starting from the smoothieboard schematics instead?
https://github.com/Smoothieware/Smoothieboard/blob/master/smoothieboard-5driver-schematic.pdf

sphawes, 2020-05-15:
Put a RAMPS wiring diagram [here](https://github.com/sphawes/index/blob/master/img/rampsWiringDiagram.jpeg)! Pretty standard aside from flipping the Y and Z axis (so that each Y axis stepper has a connector). The only real things I changed in Marlin were setting correct E-steps, picking the appropriate hardware, and flipping the Y and Z stepper driver pins. If there's something you're still missing, let me know. In the meantime I'll close this issue.

As @ncx94 pointed out, I'm currently working on a motherboard to combine the RAMPS shield, Mega, and Piggyback into one controller, but it will continue to run Marlin and is wired up very similarly to the RAMPS setup. So far I've had zero issues getting Marlin handling OpenPnP commands. This could change as I start to ask more of it, but it's also quite easy to add functionality to Marlin, so I don't mind making firmware updates to facilitate integration with OpenPnP.

closed 2020-05-15 by sphawes