opulo-inc/lumenpnp #11: Change bottom rollers for eccentric nut

Frzoen, 2020-05-16:
Change design of the carriages to use eccentric nut.
ie. this [Eccentric Spacers - Amazon](https://www.amazon.com/PSCCO-Eccentric-Spacers-Wheel-Printer/dp/B083YTJ7S3/ref=sr_1_2?dchild=1&keywords=3d+printer+eccentric+spacer&qid=1589662539&sr=8-2)

This will help in tightening the carriage rollers onto aluminium profiles.
This is also opportunity to change the design to use one less roller/carriage by changing square pattern to triangle. This shouldn't impact stability of carriage and help tightening BOM.

sphawes, 2020-05-16:
I love these things, but I'm concerned about their usage here. Instead of being inserted into a hole on a nice piece of steel, they'd be going into a hole on an FDM part. I'm not currently super happy with the current slot method, so definitely looking for an alternative, but these in a printed part might end up causing problems. Are there other types of tensioning we could use for these rollers? Or maybe some way we could have these eccentric spacers slot into something a bit more substantial?

Frzoen, 2020-05-16:
Just finished design for eccentric nut. 

I will print it and test with one salvaged from my printer. Will get in touch soon with test resoults

[image]

Frzoen, 2020-05-16:
Maybe mounting mechanism similar to this from EleksMaker featured in Marco Reps's videos?
[mentioned video](https://youtu.be/1hFNj86L7sk?t=120) (mind the goat).

There is sheet of acrylic... maybe 3d print will be similar in strength.

Frzoen, 2020-05-17:
Ok, so after first test that looks promising. 

For now i cannot test it with stepper motor but i tried to break it using a lot of force and taking advantage of the lever with 2020 profile and I could't break it. 

Right now i will adjust the height of the spacers for the v-slot rollers.

[image]
[image]

G-Pereira, 2021-02-14:
As discussed this change would require increased maintenance since the nuts would loosen over time and also not the best to use against FDM parts.

closed 2021-02-14 by G-Pereira