opulo-inc/lumenpnp #13: Front feeder rail too far -Y axis. Camera cannot see pick location.

sphawes, 2020-05-16:
The front feeder rail is too far in the -Y axis (towards the user). Shifting it back 20mm will allow the nozzle and the camera to easily see the pick area. This will also likely allow a wider range of other feeder designs to work with pre-pick vision.

Frzoen, 2020-05-17:
What about shifting feeders in X-axis apart to accommodate space for camera between them? Then camera could be even closer in Y axis

raouldeheer, 2020-07-28:
Wassen't this issue fixed in cad _v1.1.0_?

sphawes, 2020-08-02:
v1.1.0 shifted the feeder rail forward by 20mm, but it has yet to be confirmed that the camera can see the feeder output. this depends on 1) the feeder design, which is still in flux, and 2) new camera mount for the head, which is being designed. once it's confirmed it's visible ill close this up.

daveismith, 2021-02-04:
@sphawes is this still relevant with the latest feeder and free cad designs?

G-Pereira, 2021-02-04:
Yes @daveismith this was fixed!

closed 2021-02-04 by daveismith