opulo-inc/lumenpnp #14: No assembly instructions.

Frzoen, 2020-05-17:
There is no assembly instruction for the mechanical part of project

arturo182, 2020-05-31:
I'm currently in the process of printing the 3d printed parts and should be receiving most of the mechanical in the coming week. I will try to take lots of photos and document my assembly attempt (hopefully aided by @sphawes), we could use that for the instructions later :)

arturo182, 2020-06-04:
You can see my Twitter thread (https://twitter.com/arturo182/status/1268657313696698370) for some progress photos, I also provided the photos to Stephen if he wants to use them in the instructions.

Frzoen, 2020-06-06:
That would be great. As from my side i could add exploded views from cad. Unfortunately i didn't took any picture during assembly...

LoneWalkerWolf, 2020-08-15:
I too am going to she documenting my assembly of the Index. I will probably upload a .zip file with all the pictures I took. I am also planning to record a video of my assembling the machine.

daveismith, 2021-01-28:
We've got a google doc with assembly instructions currently. It's in the code lab format for now but that's under discussion as a few other formats have been suggested. Once content is complete we will convert to markdown and add to the repo with the intent to generate a site which could be hosted using GitHub pages.

meutzitzu, 2021-02-09:
one can consider using TechDrawWB and ExplodedAssemblyWB in FreeCAD to generate instructions.

G-Pereira, 2021-02-10:
> one can consider using TechDrawWB and ExplodedAssemblyWB in FreeCAD to generate instruction illustrations

That is in our plans as well as animated assemblies using FreeCAD rendering capabilities using Cycles (Blender) integration!

G-Pereira, 2021-03-08:
Thanks @Frzoen ! The assembly instructions are now present in the assembly-guide branch

closed 2021-03-08 by G-Pereira