opulo-inc/lumenpnp #16: Suggestion to change nema11 hollow shaft to a different system for angular positioning

Arthur-Silveira, 2020-05-31:
I was thinking about changing the Nema11 with hollow shaft to a different angular positioning system.
It seems to be very difficult to find such a motor here in Brazil, so I'm planning on using the same standard NEMA17 used on other axes.
I will try the approach on this drawings (really sorry for the ultra low quality on hand drawing, absolutely nothing is on scale).

Any thoughts to improve this idea??

[image]

Arthur-Silveira, 2020-05-31:
Nevermind for this suggestion.
I'll have to import the nozzle and nozzle adapter anyways, so I will just add the motor to the list as well.

I'm thinking about this kit from Aliexpress instead of the RobotDigg suggested on the BOM

https://www.aliexpress.com/item/4000090366591.html?spm=a2g0o.detail.1000014.19.a7b459d0gMMFR0&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.13338.170725.0&scm_id=1007.13338.170725.0&scm-url=1007.13338.170725.0&pvid=1c21acfd-dd72-44ef-a7ba-3e56ca9c6147&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.13338.170725.0,pvid:1c21acfd-dd72-44ef-a7ba-3e56ca9c6147,tpp_buckets:668%230%23131923%2316_668%23808%234093%23765_668%23888%233325%2315_3338%230%23170725%230_668%232846%238111%23469_668%232717%237565%23778

bjonesrock, 2020-06-01:
I'm working on putting together a pnp as well, and didn't want to buy the RobotDigg kit either. I bought NEMA8 steppers with hollow shafts, and juki nozzles from ebay and then bought quick connect fittings with 1/4" NPT threads. 

I printed a cap that screws into one side of the NEMA8 motor with M2x20 screws and then the quick connector fitting screws into the cap. Super cheap and simple way to replace the swivel adapter with a common part and a 3d printed part.

I also printed an adapter to connect the 5mm hollow NEMA8 shaft to the 7mm Juki shaft, but it's not perfectly centered, which I don't like.

Here are some photos and the STEP files for the NEMA8 cap and the Juki adapter.
[image]
[image]
[image]
[image]
[juki-nema8-connectors.zip](https://github.com/sphawes/index/files/4711643/juki-nema8-connectors.zip)

Arthur-Silveira, 2020-06-02:
Hi bjonesrock, 

> I printed a cap that screws into one side of the NEMA8 motor with M2x20 screws and then the quick connector fitting screws into the cap. Super cheap and simple way to replace the swivel adapter with a common part and a 3d printed part.

Printing that cap seems to be a very good low cost option to replace the swivel connector. It might add some vacuum losses due to air leaking through the motor, but not sure if enough to affect operation.
I will definetely give a try to your idea before purchasing the Aliexpress one. I think I might find a similar motor here. This way i will only have to buy the nozzle from ebay or Ali.

> I also printed an adapter to connect the 5mm hollow NEMA8 shaft to the 7mm Juki shaft, but it's not perfectly centered, which I don't like.

Yeah, I'm very concerned about the angular repeatability due to the misalignment. I've seen someone complaining about this on that aliexpress solution I've posted the link previously as well (posting that it would make not viable to place even 0603 components, even using the original Juki adapter).

I was wondering if we could add extra support with a ball bearing on one of those positions (see picture below). The problem on the extra support on the red arrow is that it will make very difficult to change the nozzle, so probably on the blue arrow is better.

[image]

bjonesrock, 2020-06-02:
Additional support would probably help, but it's not that long. The hollow end of the stepper shaft is basically right up against the back of the juki nozzle. I wonder if I used a resin printer if the tolerances would be better. I feel like the coupler just isn't straight enough.

Arthur-Silveira, 2020-06-03:
I see your coupler is quite simple. I though it originally had more inner details to fit the nozzle.

Have you considered just printing a simple cylinder with a single and slightly smaller hole (say 4mm) in the middle, then drill the entire piece for 5mm , and then backdrill only partially for the bigger  (7mm) hole for the nozzle?
I believe it might reduce the centering misalignment (of course, depending of the bit drill and driller quality).
Or if you have access to a turning machine, even better just machine the entire part from a solid piece of nylon, of course.

Maybe another possible try is to use a commercial 5x7 shaft coupling and then cover it with some tape to keep vaccum inside.
(like https://pt.aliexpress.com/item/32826916374.html)

bjonesrock, 2020-06-03:
All good ideas. No I haven't tried just drilling it out, I'll give that a go tomorrow and see how it works. I wish I had a lathe, though I haven't had many good reasons to get one.

Hmm those flexible adapters have slots cut out which would need to be covered with some electrical tape. I just bought a pair from Amazon and will test it out

Thanks!

bjonesrock, 2020-06-03:
I tried printing it with slightly smaller holes and then drilling it with M5 and M7 drill bits on my drill press and it's not any better.

I also then tried drilling a small piece of wood with the M5 bit, and then in the same hole drilling halfway through with an M7 bit and that wasn't any better either.

The 5mm to 7mm coupler comes in the mail tomorrow and I'll let you know how that goes.

I haven't used Open PNP before, but using the bottom camera could it compensate for an off center rotation with the upward facing camera?

plasticuser, 2020-06-04:
I wonder why NEMA 11 was chosen to replace NEMA 8? Is the NEMA 8 underpowered? It has the same accuracy, is lighter and draws less power than the 11. It’s a bit quicker too.

Mostly, it’s available from many more outlets in the hollow shaft variant. 

It would be helpful to know the reason behind the decision, so we can buy the right parts and move on.

bjonesrock, 2020-06-05:
Yeah NEMA 8 motors make way more sense to me and seems to be what most people use. The flexible adapters aren't coming until Saturday so I tried adding 3 screws around the 3d printed coupler to see if I could straighten it out. But it's still not great.
[image]

plasticuser, 2020-06-05:
The NEMA 8 are a fair bit lighter than the NEMA 11 versions, so I suspect the carriage lift/return would manage much better acceleration.

bjonesrock, 2020-06-08:
[image]

I finally got the 5mm to 7mm coupler in the mail. 

It works great! It spins nice and centered. You just have to put some pressure down on the coupler as you tighten it so everything stays nice and square

Here is a link to the coupler I bought off Amazon https://www.amazon.com/dp/B0816JQBYZ/ref=cm_sw_r_other_apa_i_3zS3EbKDBYFMK

bjonesrock, 2020-06-08:
Uhh actually scratch that. The flexible coupler is solid in the center. I tried drilling a hole in the rubber plastic center and putting electrical tape around the coupler but when I blow through the hollow motor shaft the air escapes through the sides of the coupler

[image]
[image]

Arthur-Silveira, 2020-06-17:
> Uhh actually scratch that. The flexible coupler is solid in the center. I tried drilling a hole in the rubber plastic center and putting electrical tape around the coupler but when I blow through the hollow motor shaft the air escapes through the sides of the coupler
> 

Maybe using this kind of commercial coupler would be beter to avoid leakage. Some tape over the grooves (even over screws) seems to be enough.
[image]

Unfortunately, I've not yet managed to find a suitable hollow shaft motor here in BR.
For now, I will wait for the complete kit I've purchased on Ali (https://pt.aliexpress.com/item/4000090366591.html) as amazon is not an option for me. 
If anyone else find suitable options on Chinese vendors, that would be welcome.

Also, hoping the bottom camera could adequately compensate for some deviations due to misalignment.

bjonesrock, 2020-06-17:
Yeah that style adapter seems like it might work better. I tried another 3D printed adapter that has 2 flanges that an M2 screw goes into to pull the coupler tight, but it was slightly off center as well.
[image]

bjonesrock, 2020-06-22:
I feel like maybe the shaft of my motor is slightly bent. I tried ordering a metal 5mm to 7mm coupler and it's pretty good but still has a slight offset to it. I've never used open PNP before but I really hope it can self correct with the upward facing camera. If it can I think a 3D printed coupler would be fine
[image]

plasticuser, 2020-06-22:
The shaft on the end with the power connector would be the "top" of the stepper. The shaft at the other end would be the side the head would mount.

If the shaft is slightly bent on the top it doesn't matter - it's only from the top bearing down that it would introduce any eccentricity to the bottom. You could test this with a long threaded bolt (is it threaded M3?) and turning the motor and seeing if it has any eccentricity.

bjonesrock, 2020-06-22:
So I tested all the couplers on the other side of the NEMA8 motor and the metal coupler I just got is perfectly centered. 

I'll just have redesign the 3D printed vacuum hose cap to work on the side of the motor with the plug

Thanks for the suggestion plasticuser!
[image]

plasticuser, 2020-06-22:
Excellent. Glad to be of help.

bjonesrock, 2020-06-22:
If other people are interested in the metal coupler I used here is a link to where I got them:
[https://www.ebay.com/itm/2-pcs-Aluminum-Blue-Shaft-Coupling-Rigid-Coupler-Motor-Connector-CNC-DIY/353020805184?ssPageName=STRK%3AMEBIDX%3AIT&var=622244095203&_trksid=p2060353.m2749.l2649](https://www.ebay.com/itm/2-pcs-Aluminum-Blue-Shaft-Coupling-Rigid-Coupler-Motor-Connector-CNC-DIY/353020805184?ssPageName=STRK%3AMEBIDX%3AIT&var=622244095203&_trksid=p2060353.m2749.l2649)

And here are links to where I bought my Juki nozzles:
[https://www.ebay.com/itm/2-pcs-Aluminum-Blue-Shaft-Coupling-Rigid-Coupler-Motor-Connector-CNC-DIY/353020805184?ssPageName=STRK%3AMEBIDX%3AIT&var=622244095203&_trksid=p2060353.m2749.l2649](https://www.ebay.com/itm/SMT-JUKI-Nozzle-503-Type-JUKI2000-for-Component-Placement-3-1-x-1-6cm/152928513348?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m1438.l2649)

[https://www.ebay.com/itm/SMT-JUKI-Nozzle-504-Type-JUKI-2000-for-Component-Placement-L31xD16mm/152928514640?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m1438.l2649](https://www.ebay.com/itm/SMT-JUKI-Nozzle-504-Type-JUKI-2000-for-Component-Placement-L31xD16mm/152928514640?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m1438.l2649)

Here is a link to the push to connect fittings I bought:
[https://www.amazon.com/PC-04-04-Straight-Connector-Connect-Pneumatic/dp/B07L1FVKPJ/ref=sr_1_36?crid=2ESKBO2U9PBFP&dchild=1&keywords=push+to+connect+air+fittings+1%2F4&qid=1592865097&sprefix=push+to+connect%2Caps%2C187&sr=8-36](https://www.amazon.com/PC-04-04-Straight-Connector-Connect-Pneumatic/dp/B07L1FVKPJ/ref=sr_1_36?crid=2ESKBO2U9PBFP&dchild=1&keywords=push+to+connect+air+fittings+1%2F4&qid=1592865097&sprefix=push+to+connect%2Caps%2C187&sr=8-36)

And here is a link to the NEMA8 motors I bought:
[https://www.ebay.com/itm/Nema-8-hollow-Shaft-Stepper-Motor-0-6A-30mm-for-Mill-Cut-CNC-Engraving-Machine/322618173920?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2649](https://www.ebay.com/itm/Nema-8-hollow-Shaft-Stepper-Motor-0-6A-30mm-for-Mill-Cut-CNC-Engraving-Machine/322618173920?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2649)

I did the math and the component cost is about $23 less per pick and place head than what @sphawes picked out. Using 3D printed parts and common off the shelf parts makes it easier to source and not rely on a single vendor like Robotdigg.

Plus NEMA8 motors are smaller and lighter so you might be able to drive the PNP a bit faster without losing steps.

When I have a few minutes 3D model and post an updated step file of the vacuum cap adapter that fits the top of the NEMA8 motor

bjonesrock, 2020-06-25:
Alright so I made a new version of the NEMA8 quick connect hose adapter for the vacuum. This time it attaches to the top side which has the plug. I cut away at the adapter for easier access to plug in the motor. The threads for the M2 screws on the NEMA8 motor on the bottom side of the motor which means the 20mm screws I had are too short, I'll order some M2 35mm screws which should do the trick. I'm attaching the STL and STEP files for this adapter

[VacuumCapNEMA8Top.zip](https://github.com/sphawes/index/files/4832757/VacuumCapNEMA8Top.zip)

[image]
[image]
[image]

bjonesrock, 2020-06-29:
I got the m2 30mm screws in the mail and screwed the cap on the motor, but there is some air leak between the 3D print and the motor. I'm thinking of using some petroleum jelly between the two pieces to make it airtight. If that doesn't work I could add a small groove in the print for a rubber o-ring. Any thoughts on how to seal it up? 

I've been working on how I want to control the Z axis on my PNP and I think I'm going to use a GT2 belt. So I added mounts to the 3D printed cap to screw into a short piece of linear rail and also a clip to hold onto the GT2 belt.

[image]
[image]
[image]
[image]

Arthur-Silveira, 2020-06-29:
That looks great. I have some ideas

Regarding the sealing between motor and 3D printed, It would help if you send a photo of the motor "opened"to check the regularity of the surface between the motor and the 3D printed part. If it is "regular" or smooth, like the picture you sent of the bottom side (wrong side) like this image below:

> Here are some photos and the STEP files for the NEMA8 cap and the Juki adapter.
> [image]

then I would recommend a commercial oring seal (you could also make a partial groove to fit the Oring on the printed part.)

If it is irregular, like square, uneven or with sharp edges, maybe a good option would be some paste or clay like material.
Here in Brazil we have this material that is very commonly used to seal some low pressure pipings (usually the sewer pipings, leaking sinks or faucets).
Good thing about those products is that they remain "soft" and do not dry out (I mean they do not dry and harden like epoxy based, but they remain elastic).
[image]

Bad thing is that when I tried to translate it to english (caulk or caulking), all images I found seems products that will cure, dry or harden. So I'm not really sure if it is common in the US or Europe.

bjonesrock, 2020-06-29:
Yeah that might do the trick. In the US we call that Plumber's Putty. I've got some plasticine clay on hand, maybe that'll work. I had to rotate the part in order to print it without supports so it has lots of layer lines which makes sense why it wouldn't seal very well.

[image]

Arthur-Silveira, 2020-06-30:
If it is also leaking between the layers even with finer layer steps (0.1 mm layer for example) maybe some paint or coating might help reducing the leakage. Maybe some spray painting, or nail polish ? Never tried painting PLA before.. there might be some preferred options that best adhere to PLA plastic.

This seems very good, but very specific though 
[image]
 https://3druck.com/3d-druckmaterialien/xtc-3d-verleiht-3d-gedruckten-objekten-eine-ebenmaessigere-oberflaeche-3028167/

I would give a try with nail polish + commercial Oring (to avoid the mess with the Plumber's putty) as 1st option, since they seem commercially available everywhere.

bjonesrock, 2020-06-30:
I put a thin layer of clay between the 3D print and the NEMA8 motor and that sealed it really well :)

Plumber's putty is pretty common, but clay works just as well. Smoothing the part might work, but PLA is harder to smooth.

Since I rotated the part to 3D print it the tolerance on the push to connect adapter was a little loose so I wrapped some more teflon tape. I'll tighten up the part and try printing it again and upload the file soon

[image]
[image]
[image]

bjonesrock, 2020-07-01:
Alright I tightened the hole that the push to connect fittings screw into so you don't have to seal it with extra teflon tape. Here are the updated STEP and STL files 
[VacuumCapNEMA8TopGT2BeltTight.zip](https://github.com/sphawes/index/files/4859855/VacuumCapNEMA8TopGT2BeltTight.zip)
[image]

Arthur-Silveira, 2020-07-24:
My motor+nozzle+adapter kit finally arrived from China.
Mi kit also came with nozzle connected to what seems to be the wrong side, but first, I'm having some electrical issues.

@bjonesrock can you share the electrical connections of your motor?  (it seems to be similar to mine). Are you using 12V as well?

My little motor is pulling `ALONE` over 0,8 A (@6 V only, I've reduced the voltage because my PSU was triggering overcurent). It has to be something about wiring. When I replace this one with two NEMA17, current keeps around 0,6 A

I've measured continuity between `RED to BLACK` and also `YELLOW to GREEN`, so I`ve switched the terminals at RAMPS side (the connector at the motor is unchanged as arrived).
I've measured about 7 ohms each windings (that is about twice my standard NEMA 17 motors that have about 3.something ohms), so it is probably not the motor

[image]
[image]

wjsteele, 2020-07-24:
Did you adjust the trimpots to set the proper vRef?  See here if you haven't: [Pololu A4988](https://www.pololu.com/product/1182)

Arthur-Silveira, 2020-07-24:
> Did you adjust the trimpots to set the proper vRef? See here if you haven't: [Pololu A4988](https://www.pololu.com/product/1182)

I had not, thank you very much for the link.  But still wondering why I was getting lower current ratings on a motor with lower resistance on same pot. adjustment.

I don't have the motor specs, but I've put coil current to about 350 mA (about 200 mA at PSU). If I adjust to less than that, the motor does not spin right. With that current, it is spinning very smootlhy
It is getting a little warm though, but seems OK (no smell yet).

Now, I'll try to detach that coupler to place it on the other side of the shaft. 
It is very misaligned: https://drive.google.com/file/d/1tLsLkWzka8an-K4w2UKLxB3A6DuTmg_x/view?usp=sharing

daveismith, 2021-02-04:
This seems stale and like the project has moved on. Should this be closed?

G-Pereira, 2021-02-04:
Indeed. NEMA 11 were picked and work best for microstepping and also have more thermal mass.
I also think we can close this @daveismith

closed 2021-05-14 by G-Pereira