opulo-inc/lumenpnp #17: Reason for using RAMPS over Smoothieboard?

plasticuser, 2020-06-02:
Is the $70ish difference in price the only reason for going from the well OpenPnP supported Smoothieboard 5X to the clunky and so-so RAMPS board?

I've seen that Smoothieboards seem to operate steppers much more smoothly with much more elegant acceleration curves.

Or is that just me?

DurandA, 2020-06-02:
I think you will be interested by [this comment](https://github.com/sphawes/index/issues/9#issuecomment-628983428) from @sphawes:

> As @ncx94 pointed out, I'm currently working on a motherboard to combine the RAMPS shield, Mega, and Piggyback into one controller, but it will continue to run Marlin and is wired up very similarly to the RAMPS setup. So far I've had zero issues getting Marlin handling OpenPnP commands. This could change as I start to ask more of it, but it's also quite easy to add functionality to Marlin, so I don't mind making firmware updates to facilitate integration with OpenPnP.

plasticuser, 2020-06-02:
RAMPS is economical, but so, so ugly. I don't think designing it and the Arduino onto a common PCB improves anything. All other things being equal (it's a CPU G-Code to stepper controller and feedback system after all) I think I'd take the 120MHz Smoothieboard over the 16MHz ATmega2560...

Mostly, I'm concerned about future support. I'd prefer to be in the middle of the mainstream on the software-touching side as far as OpenPnP is concerned.

Am I overthinking it?

sphawes, 2020-08-02:
The new mobo design is still running marlin, but now runs on an STM32F407. should be plenty to handle what's being asked of it, and as @DurandA mentioned, I'm happy to invest the time in updating Marlin to keep up with the demands of the machine (of which there wont be many extra, i believe).

closed 2020-08-02 by sphawes