opulo-inc/lumenpnp #18: 20x20 rail too small to rigidly support carriage

plasticuser, 2020-06-03:
During work the carriage can apply enough force to the rail to twist on it through the metal having very slight flex, but more from the lash in the bearings and touching surfaces.

This won't be as much of a problem near the end of the rails, where just the carriage lash exhibits, but towards the middle of the rail the effect will be enough to cause the head to shift position measurably. These uneven-by-position effects are the hardest to calibrate out in software.

The effect would worsen over time with wear, and also become more apparent if people build wider machines.

I suspect going to 20w x 40h rail for that one rail would reduce this effect by 75% or more and reduce wear on the rollers considerably. Remember, that rail is sitting on 2x 20x20 rails, one each end, so each of those rails is only carrying half the load, and the load is centered on those rails.

Arthur-Silveira, 2020-06-03:
I've not started assembling it yet, but i agree and I'm also concerned about this . 
I've already purchased one exta 20x40 profiles as well for the center axis thinking about this. 
Aside from the higher load, the center rail rollers (X axis) will also suffer from torsion effect due to more weight on one side. That does not seem to be the case on the Y axis. And as you already mentioned, they will share the load and carry half the weight each side. Using a wider profile seems to reduce the torque and stress on the rollers as well.

Also, some extra supports connecting with the bottom rails might help to distribute the load even more (like in this [updated picture](https://github.com/sphawes/index/pull/5) by [bennymeg](https://github.com/bennymeg)) on the Y axis. 

It seems to be a good idea for me to go with the 20x40 profile on that single X axis (but its a simple guess for now, more impressions only after assembly)

plasticuser, 2020-06-03:
Doubling the beam height halves the leverage during picking, placing, tool changes....the whole time. This can only decrease slop and reduce wear. It also reduces the carriage's tendency to want to rotate in other planes too when moving, which will allow a little higher driven speed without vibration possibly bumping parts off.

sphawes, 2020-08-02:
My testing thus far has not shown that the 20x20 rail is too small to support the x gantry. If you try building with a 20x20 size rail and notice issues related to this rail size, please reopen this issue and we'll take a crack and changing the size. In the meantime I'll close this issue.

closed 2020-08-02 by sphawes