opulo-inc/lumenpnp #24: Just a bit of feedback and a few notes

Malossi166, 2020-06-20:
Hi, you asked in your recent YouTube video for some feedback on your pcb design.
It is impossible to build a "one fits all" pcb. You need a GPIO header to make it easy to add all kinds of boards and stuff. If you run out of GPIO pins on the Mega just use pins of things most people likely will not need anyway. Also add GND and V to this header. This way anybody can design another board/sensor/anything and connect it using just one header. If some mod or board gets really popular you can add it in a later revision.

Oh, and I do not recommend using some sort of 12v heat gun. They need a lot of power so your power supply have to be able to provide a lot more power (like 600w or even more) and you need pretty beefy, not very flexible cables to supply this amount of power (600w@12v = 50amp -> AWG8 wires) not to mention the cost of a good 800w 12v psu. I think you best option is to either just use an oven or to use the handpiece of an 858D hot air station. You can buy them for ~$12 and they even include a temp sensor. Not every paste needs the same temperature and I think it is not a great idea to use something without temperature control. This adds a lot of complexity, but I think it will be worth it as you need good solder joints to get something usable.

sphawes, 2020-06-25:
Thanks for your thoughts @Malossi166! 

yes, i totally agree, i think it's impossible to make a true "one-size-fits-all." the next revision of the motherboard will have exactly what you suggested: a ton of headers with GPIO pins, along with breakouts for SPI and I2C so folks can add on their own features. is there a standard type of expansion port that would be good to follow and would help support existing peripherals? the LCD/encoder module that fits into the RAMPS shield comes to mind.

ah yes, this was a total flub on my part. forgot about the fact 12v would take such a serious amount of current to reach the same power. I'll be putting reflow on hold indefinitely until main pnp functionality is totally working and reliable, but I'll most likely look into integrating an oven into the situation.

adlerweb, 2020-06-26:
Jepp, a connector with a pinout compatible with EXP1/EXP2 as found on [RAMP1.4 Smart adapter](https://www.thingiverse.com/thing:26613/files) or [SKR Boards](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4-SCH.pdf)(Bottom left of the CPU) might be a good idea if there is enough space. Basically just a few GPIOs and SPI to drive a [(G)LCD, SD-Card and various buttons/encoders](https://reprap.org/mediawiki/images/5/51/RRD_FULL_GRAPHIC_SMART_CONTROLER_SCHEMATIC.pdf).

Malossi166, 2020-06-26:
It is very hard for one person to think about everything and the first idea is rarely the final one @sphawes .
I think it is a good idea to put the reflow at hold for now. It makes things just more complex and it is pretty easy and cheap to get an external solution for now. It also helps to keep the weight a fair bit lower on tool head, although I am not sure what your speed target is.
A Ramps (pin)compatible header seems to be a good idea as it is likely that at least some addons already created for it can be also used for this project as they share so many things in terms of hard and software. If you like to make even more pins available, you can add an additional fully custom header.

closed 2020-08-02 by sphawes