opulo-inc/lumenpnp #25: CH340G

CrashOverrideProductions, 2020-06-21:
have you thought of using a FTDI chip for the usb to serial for the ATMega.. 

in my experience the CH340G has caused issues in some of my designs (automotive diagnostic tools) usually just poor performance and compatibility issues with MacOS, these could just be my implementation but ive found the FT230x chip is more reliable, plus at least in the land down under is cheaper than the CH340G $7 vs $3

plasticuser, 2020-06-21:
As a former FTDI user with a memory, I recall when FTDI bricked products with non-compliant (counterfeit) chips. FTDI bricked a lot of gear this way. There's a reason FTDI USB chips are cheaper than they used to be - FTDI destroyed their reputation by trying too hard to maintain it.

I use devices on my MacBook Pro daily that have the CH340G, and have incorporated it into various ATmega designs without issue. However, I go to some lengths to make sure I am buying genuine WCH parts, not counterfeits.

bjonesrock, 2020-06-21:
Yeah I second plasticiser. No one trusts FTDI anymore. There are tons of chips in the gray market and it's hard to know if you have real or fake chips. I'd highly advise against using FTDI.

CrashOverrideProductions, 2020-06-21:
FTDI Gate was a big pain for me, i had i heap of cheap gear stop working during development (obviously counterfit) but never had an issue with any of it since i started buying from mouser or digikey,  was just a thought with the issues ive had with CH340 but thats more than likley specific to what i was trying to do with it (single wire data bus with a non standard baud rate)

brainstorm, 2020-06-21:
I would actually discourage using the CH340 since it'll be EOL'd in OSX soon and there are more modern alternatives:

https://www.electronics-lab.com/ch551-1-80-usb-mini-development-board-based-8-bit-c51-micro-controller/

Perhaps even throw in USB-C while you are at it? You could even get rid of the barrel jack connector and power most of the board via USB-PD?... rabbit hole, I know, but also a bit more future-proof?

soundstorm, 2020-06-21:
I'm using 340C on my products, as I then can save the resonator or in this case crystal and capacitors.

sphawes, 2020-06-25:
the EOL on CH340 is the most worrisome part of this for me. ultimately, we'll be switching to a nice beefy ARM chip for the processor with native USB, so we can completely remove a USB <> UART bridge entirely. thanks all for the thoughts!

closed 2020-06-25 by sphawes

brainstorm, 2020-06-25:
... @sphawes or an ESP32S2 which has native USB and wifi "for free" ;)

adlerweb, 2020-06-26:
> ... ESP32S2 which has native USB and wifi "for free" ;)
...and probably the same famously bad performance for time sensitive applications as it's predecessors. Interrupt delays are a pain on their architecture. Also there would be ARM based ICs with integrated WiFi available. Since I, however, don't think @sphawes meant he would switch to a full fledged ARM SBC running Linux/OpenPNP on the machine itself, you would require cables for the webcams nevertheless and WiFi at the PC level would enable far greater control without messing with RF on a somewhat noisy Board controlling motors.

plasticuser, 2020-06-28:
I checked with the manufacturer (WCD) and they say they're pushing out updated drivers before the release of the next edition of Mac OS X. They will be fully 64-bit and use the new kext signing scheme.