opulo-inc/lumenpnp #27: Missing motor specification for feeder

richardklingler, 2020-07-06:
Hello

I don't see any specs about what motor is actually used for the feeders in the parts list nor in any description.

raouldeheer, 2020-07-16:
In [this video](https://www.youtube.com/watch?v=2foSlnGWeu0&list=PLIeJXmcg1baLBz3x0nCDqkYpKs2IWGHk4&index=6&t=0s) Stephen used a N20 Worm Gear Motor. I'm not sure which version but they're cheap so I would get the one with the most torque.
In [this video](https://www.youtube.com/watch?v=an5PXKHXsvQ&list=PLIeJXmcg1baLBz3x0nCDqkYpKs2IWGHk4&index=8) Stephen finds out that his N20 motor isn't strong enough to peel the tape. So he used an RC car motor, he doesn't says which motor it is.
I hope this information helped.

G-Pereira, 2021-03-08:
Hey @richardklingler !
Feeder motors used on our design are:
All-metal TT motors: https://www.aliexpress.com/item/33058637462.html
N20 50 rpm 6V: https://www.aliexpress.com/item/33022320164.html

We don't have assembly instructions or BOM for the feeder but we are working on it!

sphawes, 2021-10-05:
Thanks for the note @richardklingler! This is a duplicate of #63, closing.

closed 2021-10-05 by sphawes