opulo-inc/lumenpnp #31: Need a list of approved PCB vendors for feeder pcbs

TomKeddie, 2020-07-08:
JLC have reject the indexing wheel.

Order Reviewed:2020-07-08 18:00:33
Check the audit failed reason:

the corner cutout is too small, kindly fix to hit our capacity 6x6mm at leat.

adlerweb, 2020-07-08:
Hm, I guess that's up to who's checking the design. Had no problems some weeks ago. Might be worth a shot just canceling and reordering them or telling them to "just try"...

Arthur-Silveira, 2020-07-19:
I think It could also be a good option to put those boards in the "Shared Projects" at PCBWay. Since they are sponsoring the videos on Youtube, this might also help to "sponsor your sponsor". It would also help those that are not familiar with the PCB manufacturing process to avoid generation of gerber files, drill files, ODB++ files etc.

Appended: I just found 3 of the boards already there:
- Ring light rev000
- Motherboard rev000
- Motherboard rev000_1

Still missing the feeder boards and rotary encoder there.

sphawes, 2020-08-02:
@Arthur-Silveira that's a good thought. given that designs are not stable, i wont add these links quite yet, but once they're in a good state putting the "order now" link will remove the risk of a design review rejecting the board. I'll leave this open until that happens

Decee1, 2020-10-20:
+1 to no problems with ordering at JLCPCB.
They cancelled another PCB I added because it was missing drill files but the Indexing wheel came though without a problem

daveismith, 2021-02-04:
For JLC I found that the preview system wouldn't take the automated export files initially because it had an invalid drill file. I had to add two small vias to make a valid drill file.

G-Pereira, 2021-03-08:
Thanks everyone! This is now fixed on the feeder-rev05 branch

closed 2021-03-08 by G-Pereira