opulo-inc/lumenpnp #33: [CAD] - Front_left_leg added ramps endswitch holes and slot

Arthur-Silveira, 2020-07-28:
Added holes and slots to install the common `ramps endstop` board (mechanical endstop switch) on front left leg (Y zero switch). Based on @sphawes front left leg v 1.1.0
[image]

I know this board is a little more expensive than using simple endswitches, but is very common, so I think quite a few people might use it
It also has some cool leds on it, and usually comes with cables already).
I've also added some slots for PTH terminals to fit, so no need to cut them at all.
Able to use wooden screws to hold the PCB, but standard M3 thread also does the work

Had to print with supports to hold the hanging slots. 
I've printed the 1st version at 100% and had to do some hammering to fit the 20x20 extrusions in place. This time I've printed with 101% size to compensate for material (PLA) shrinkage (thanks arturo for the tip). Extrusions now still fit firmly yet, with no difficulty. 

I've though of adding some supports directly to the CAD file to support the recess, but I was worried about that big bridge on the original foot gap, and was going to print with supports anyway.
See some attached pictures. The total size of the recess was bigger than the board because I was thinking about some extra space for the connector to attach/detach, but in the end, the recess was so shallow that it was not necessary (maybe completely remove the PCB recess, leaving only holes and slots)

I ended on removing @sphawes original endswitch holes, but maybe a mixed solution with both options might add some flexibility on final solution.

STEP FILE attached into a zip file
[IndexPnP_1.1.0b_front-left-leg_ramps_endswitch v5.zip](https://github.com/sphawes/index/files/4989773/IndexPnP_1.1.0b_front-left-leg_ramps_endswitch.v5.zip)

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

sphawes, 2020-08-02:
Thanks for the awesome update @Arthur-Silveira! You have a really good point here, most people already have this board. Plus it's a bit cleaner to have an actual PCB to mount to the frame with a proper connector, rather than just the bare component and soldered leads. I'll be working on cad for v1.1.1 soon, ill make sure to integrate this in! Soon I'll be switching all the cad over to FreeCAD so that folks can edit the source files in Github, but in the meantime I've just been making edits myself in Fusion based on suggestions. I'll leave this open until I roll in this feature.

Arthur-Silveira, 2020-08-04:
Hey @sphawes I'm glad I could provide a little help. 

Updating this issue, I had to modify one of the belt clips to improve contact to Y endswitch. It was missing the X gantry (too close to the edge), so I've added a projection to it, see attached pictures. Not it has a nice flat surface to hit.
[image]
[image] [image]

Here is the clip in Fusion and step formats, if it helps
[IndexPnP_1.1.0_belt-clamp_endsw v5.zip](https://github.com/sphawes/index/files/5023962/IndexPnP_1.1.0_belt-clamp_endsw.v5.zip)

And as for the X axis endstop switch, I fixed it on top of the left gantry instead of the side. The length of the PCB matches almost perfectly. I've just applied the same holes and slots profile, and also had to reduce the top edge chamfer to put the screw holes closer to the edge (i think original chamfer was 2.5 mm, I had to reduce it to 1 mm.
[image]
[image]
[image]

In this case I'm not uploading the CAD file for this model because I've made another modification to it to fit different rollers (original V-slot rollers are still coming from China, and meanwhile I'm using those white rounded rollers. ). They have different dimensions, and seem to perform worse. I definitely do not recommend using them. I just did because I didn't want to wait.
[image]

Best regards,

daveismith, 2021-02-04:
This is in place in the free cad versions. Can this be closed?

G-Pereira, 2021-03-08:
I just tested their assembly and they are ok!

closed 2021-03-08 by G-Pereira