opulo-inc/lumenpnp #36: [CAD] - Suggestion to improve X gantry front (increase frontal face by 2mm)

Arthur-Silveira, 2020-08-06:
I suggest, to add extra 2 mm on the frontal face of the front X gantry.

The M5x45 screw I've used to hold the rollers have a 5mm lenght head, and some of the head remains exxposed, and will interfere with the linear rail (see pictures).
[image]
[image]
[image]

The head top is 1.35 mm above the surface. Those extra 2 mm in the frontal face would completely cover the screw.
[image]

There is 3.8mm of exposed thread in the back nut (I'm using a simple nut). I know that the nylon lock nut will be a bit longer, but I think it isstill enough to spare for increasing to the front face depth.
[image]

This is a suggestion. If not accepted, maybe adding a warning on the BOM to use M5x45 screws with head length smaller than 3.6 mm length to avoid this issue.

OJFord, 2020-08-14:
5mm head height is per ISO4762 according to https://www.engineersedge.com/iso_socket_head_screw.htm

LoneWalkerWolf, 2020-08-15:
To those who are having the same problem:

I have made the changes to this STL File, please use this to reprint your x-gantry-front. 
https://drive.google.com/file/d/1r3FPwXkL4W2Xfw1klLjvMGTdvRtRaKHC/view?usp=sharing

Thanks.

G-Pereira, 2021-01-28:
This has been incorporated to the project! Nice work!

closed 2021-01-28 by G-Pereira