opulo-inc/lumenpnp #43: KiCad Sch & Board to PDF

paloky, 2020-10-07:
Hi Stephen.
Can you include in this repository all KiCad documents in PDF.

It's better for view more fast. (And if don't working with KiCad)   ;-)

Thank a lot.

LoneWalkerWolf, 2020-11-01:
@paloky Here you go. **Please do however note, that this was the motherboard (as of 1/11/2020).** For future reference, please download KiCad for your convenience. (https://kicad.org/)
[moboPDF.zip](https://github.com/sphawes/index/files/5470998/moboPDF.zip)

G-Pereira, 2020-12-30:
It would be nice to use [KiBot](https://github.com/INTI-CMNB/KiBot)!

daveismith, 2020-12-31:
I'm working on this with github actions and KiBot. I'm planning to output:

- ibom format
- csv bom
- html bom
- schematic
- pcb layer pdf (one layer per page, coper first, then silk)
- gerbers (compatible with PCBWay, JLCPCB, DirtyPCB, OSHPark & Aisler)
- drill file
- board top render
- board bottom render
- pick and place file
- JLCPCB compatible BOM (LCSC Part Numbers included)
- JLCPCB CPL file
- Step File

daveismith, 2021-01-28:
This has been completed in the feeder-rev05 branch with #114 but still needs to be done for the mobo-rev03 branch.

Take a look for an exports directory under each board directory.

daveismith, 2021-02-04:
This is supported in both the mobo-rev03 and feeder-rev05 branches. Look for an export folder inside the folder containing the schematic and board file.

PR #126 (2021-02-01): Feature/43 mobo autogen
- Adding github action to generate outputs automatically
- Adding BOM information to schematics
- Adding some NC flags to clean up a few ERC issues.
- Adding hash information on output.

Closes #43 and #99
closed 2021-02-04 by daveismith