opulo-inc/lumenpnp #48: [CAD] - Z Axis Endstop for OpenPnP alignment.

LoneWalkerWolf, 2020-11-01:
After trying to get OpenPnP to accurately zero the Z axis, I think it might be necessary to add an extra endstop on the axis. I have designed a quick mount for one (attached below) and this has proven to work on my machine. _(please bear in mind I am still using a RAMPS 1.4 board)_

[image]
[image]

Here are the files: [z-axis-endstop.zip](https://github.com/sphawes/index/files/5471252/z-axis-endstop.zip)

Cheers.

daveismith, 2021-02-04:
The latest free cad designs have this in place at the top right of the Z axis part. Is this sufficient to close the ticket?

G-Pereira, 2021-03-08:
Closing this since it is indeed in place and no answer from the author

closed 2021-03-08 by G-Pereira