opulo-inc/lumenpnp #55: Exchanging vacuum sensor analog signal processing with HX711

sphawes, 2020-12-04:
We currently use two opamps and an instrumentation amplifier to process the signal from the vacuum sensor. This signal is then sent to an ADC pin on the STM32F4. These parts are expensive, and it is difficult and less than ideal to implement reading from an ADC in Marlin. 

The [HX711](https://cdn.sparkfun.com/datasheets/Sensors/ForceFlex/hx711_english.pdf) is an ASIC designed for processing signals from load cell Wheatstone Bridges (which is effectively what [our vacuum sensor](https://omronfs.omron.com/en_US/ecb/products/pdf/en-2smpp-03.pdf) is). The HX711 also provides a digital interface for this data, so no ADC usage is necessary.

It's unknown if this switch will work; some investigation with the chip and how difficult it is to implement into Marlin still needs to be investigated.

martinbogo, 2020-12-21:
I'm going to buy an HX711 and probe and poke at it

G-Pereira, 2020-12-31:
I can work out the schematic!

G-Pereira, 2021-01-02:
[English datasheet](https://cdn.sparkfun.com/datasheets/Sensors/ForceFlex/hx711_english.pdf)

G-Pereira, 2021-02-14:
As discussed in Discord we will use a general purpose ADC that would fit best our sensor since HX711 is designed for load cells.
In comparison to HX711 a general purpose ADC:
- Allows linearize output specific to our type of sensor
- Easier to implement in software
- More valuable contribution to marlin (external ADC integration than specific HX711 code)
- More flexible (gains, signal conditioning)

closed 2021-02-14 by G-Pereira