opulo-inc/lumenpnp #56: Mobo Cover

sphawes, 2020-12-04:
The motherboard is fairly exposed when mounted to the bottom of the panel. A snap-on cover would be a great optional print for folks to have if they want to protect their motherboard.

blucky10, 2021-06-05:
Is someone working on this, If not I would like to contribute.

G-Pereira, 2021-06-07:
@blucky10  No one is working on it, feel free to start! Be sure you use FreeCAD being careful with topology references, watch [here](https://www.youtube.com/watch?v=e0pvw1weSuM)

blucky10, 2021-06-09:
Can I get step file of Mother Board, with all the connectors to design the cut outs for them. As I have difficulty in obtaining the proper 3d files of the connectors.

[image]
[image]

blucky10, 2021-06-13:
@G-Pereira , I'm ready with my initial design, thanks for the link, 
Just want to discuss some design consideration, will need a call I Guess. 

[image]

G-Pereira, 2021-06-14:
Hello @blucky10 it looks nice but the new Mobo rev03 changed the board dimensions, now it's squared. The board isn't released yet but it is nearly ready.
Here it is the [step file](https://github.com/index-machines/index/blob/mobo-rev03/pnp/pcb/mobo/export/mobo-3D.step)

G-Pereira, 2021-06-14:
@blucky10 are you on our Discord? We can have a call if you want!

blucky10, 2021-06-14:
> @blucky10 are you on our Discord? We can have a call if you want!

Yes I'm on discord,  just messaged you please check

Magpie-81, 2022-08-22:
I made a very simple protection cover for the Rev3 MOBO. It's available at Printables 
https://www.printables.com/model/242429-lumenpnp-mobo-protection-plate

closed 2022-09-16 by sphawes