opulo-inc/lumenpnp #57: Mobo has most/all SMT footprints, and hybrid footprints for both SMT and THT where possible

sphawes, 2020-12-04:
The REV02 motherboard design has a number of THT footprints that cannot be assembled using a pick and place. As many footprints as possible should be switched over to SMT. Some footprints (namely stepper driver headers, and most connectors) should be custom, hybrid footprints that allow for both types of parts to aid with hand-soldering, and also supporting being pick and placed.

G-Pereira, 2021-07-05:
Use: https://lcsc.com/product-detail/USB-Connectors_Jing-Extension-of-the-Electronic-Co-AF10-0-Cparagraph-Front-two-feet-Have-side-iron-LCP-Vinyl-6-5-SMT-High-temperature_C39429.html

sphawes, 2021-09-20:
going to close this! board space-wise, and cost of component wise, this doesnt make sense to do. could be worth revisiting in the future, but going to close for now.

closed 2021-09-20 by sphawes