opulo-inc/lumenpnp #59: Decide and implement ID method for feeders

sphawes, 2020-12-04:
The feeders need some way of knowing what slot they're in. The current method uses a voltage divider on the feeder floor, but the chance of voltage drop across the connector makes this less than ideal for reliability. 

There are quite a few ways to do this, but a quite good proposed solution is using one ID resistor on the feeder floor PCB, and using that in conjunction with a resistor on the feeder building a voltage divider. Wiring it this way will help remove voltage drop across the connector as a factor. A wiring diagram is below, assuming an RJ45 connector:
[image]

This needs to be tested to ensure it works reliably. Resistor values and subsequent ADC readings also need to be calculated for 32 discrete IDs. If this does not work reliably, another method should be identified.

bholmes451, 2020-12-04:
What happened to the idea of using a DIP Switch Package? You can get them for less than the RJ48 port.

https://www.digikey.com/en/products/detail/cui-devices/DS01-254-S-08BE/11310937

[image]

sphawes, 2020-12-04:
hey @bholmes451!

The main reason the DIP switch is out is because it would have to be placed on the feeder itself, instead of the feeder floor. The reason for this is that there aren't enough data lines with either an RJ-45 or with the spring fingers to transfer the 5 bits of data necessary from the floor to the feeder. 

If we put them on the feeder, we're adding the extra step in the process of swapping out feeders of reconfiguring the switches, plus introduce an extra opportunity for operator error in case someone forgets to switch them. Ultimately, the one resistor on the feeder floor that does the identification costs less than the DIP switch we'd need (especially because it would have to be SMD, right angle, and ideally with piano switches to make it easier to change).

It would definitely solve the problem in a clean way, but would require more work to configure, and more cost to implement. At least, that's my thinking on it!

Fanfwe, 2020-12-04:
Why not using a DS2431 1-wire EEPROM ?
Pros:
 - Doesn't require an extra pin on the connector compared to the voltage divider method
 - Reliable. When you read an ID, you know it's good, you don't rely on precision of resistors and ADC
 - Allows for much more than just 32 IDs to be stored (for future ideas or very large builds with a lot of feeders)
Cons:
 - Need to program each feeder floor with its ID. Another approach could be that we just use the unique ID of each chip, and have a configuration file in the host that maps a unique ID to a feeder slot, this way no need to program anything, but a special step to figure out each slot unique ID is required to be performed once
 - Need to implement 1-wire protocol in software

gasmeter, 2020-12-22:
Hi
The advantage of the one wire bus products they have noise immunity and can traverse long cable runs

The DS28E05 is about the lowest cost option. 
https://datasheets.maximintegrated.com/en/ds/DS28E05.pdf

The drawback is the lower cost parts use phantom power from the bus, but that just means a little more work on sw.
They are perfect though for ID functionality and Fanfwe said.
Peter

daveismith, 2020-12-26:
There were a few different options presented on discord which could allow for automatic slot identification. Trying to capture them for the sake of discussion / review. The feeder protocol only supports 253 feeders. My recall of the options is:

1. Use an ID chip as discussed above
2. Use an ID assignment method via buttons, etc.
4. Use DIP Switches (discounted)
5. Use the analog approach, but limit the voltage dividers based on accuracy of resistors. Using a 1% resistor, you can split the range from 0-3.3V into 14 very distinguishable steps. Using a 10% resistor your usable groups becomes about half. If you use a 1% resistor, and 2 analog lines, you could detect 196 slots. Adding a third analog line will take you to 2744 possible slots. If you go to a 10% resistor, you might want to cut that to 7. That still gives you 343 possible slots that you can detect. 
6. Use a trinary approach (ie, tied to ground, tied to VCC, floating). This would require 6 data lines to identify all slots, though you could identify 243 with only 5 lines.

Note: If selecting a 1% resistor, the following table shows a possible selection set to provide 14 steps

Target Voltage  | R1 | R2 | R1_min | R1_max | R2_min | R2_max | V_min | V_max
-- | -- | -- | -- | -- | -- | -- | -- | --
0 | DNP | 1000 | #VALUE! | #VALUE! | 990 | 1010 | 0.0000 | 0.0000
0.25 | 1000000 | 82000 | 990000 | 1010000 | 81180 | 82820 | 0.2455 | 0.2548
0.5 | 56000 | 10000 | 55440 | 56560 | 9900 | 10100 | 0.4916 | 0.5085
0.75 | 51000 | 15000 | 50490 | 51510 | 14850 | 15150 | 0.7385 | 0.7617
1 | 62000 | 27000 | 61380 | 62620 | 26730 | 27270 | 0.9872 | 1.0151
1.25 | 18000 | 11000 | 17820 | 18180 | 10890 | 11110 | 1.2362 | 1.2673
1.5 | 12000 | 10000 | 11880 | 12120 | 9900 | 10100 | 1.4837 | 1.5164
1.75 | 16000 | 18000 | 15840 | 16160 | 17820 | 18180 | 1.7306 | 1.7635
2 | 13000 | 20000 | 12870 | 13130 | 19800 | 20200 | 1.9842 | 2.0157
2.25 | 5600 | 12000 | 5544 | 5656 | 11880 | 12120 | 2.2356 | 2.2643
2.5 | 24000 | 75000 | 23760 | 24240 | 74250 | 75750 | 2.4878 | 2.5121
2.75 | 2000 | 10000 | 1980 | 2020 | 9900 | 10100 | 2.7408 | 2.7591
3 | 1000 | 10000 | 990 | 1010 | 9900 | 10100 | 2.9945 | 3.0054
3.3 | 0 | 10000 | 0 | 0 | 9900 | 10100 | 3.3000 | 3.3000

daveismith, 2020-12-26:
I did some calculations on what's required to do the separation as discussed in the original ticket. My sheet is [Index_feeder_Calculations.xlsx](https://github.com/sphawes/index/files/5743316/Index_feeder_Calculations.xlsx)

The output is:

Target Voltage | R1 | R2 | R1_min | R1_max | R2_min | R2_max | V_min | V_max | Voltage from Previous | ADC Min | ADC Max | ADC Spacing From Previous
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
0 | 100000 | 1000 | 99000 | 101000 | 990 | 1010 | 0.0000 | 0.0000 |   | 0 | 0 |  
0.1 | 100000 | 3000 | 99000 | 101000 | 2970 | 3030 | 0.0943 | 0.0980 | 0.0943 | 117 | 122 | 117
0.2 | 100000 | 6200 | 99000 | 101000 | 6138 | 6262 | 0.1891 | 0.1963 | 0.0911 | 234 | 244 | 112
0.3 | 100000 | 10000 | 99000 | 101000 | 9900 | 10100 | 0.2946 | 0.3055 | 0.0983 | 365 | 380 | 121
0.4 | 100000 | 13000 | 99000 | 101000 | 12870 | 13130 | 0.3730 | 0.3864 | 0.0675 | 462 | 480 | 82
0.5 | 100000 | 18000 | 99000 | 101000 | 17820 | 18180 | 0.4949 | 0.5120 | 0.1085 | 614 | 636 | 134
0.6 | 100000 | 22000 | 99000 | 101000 | 21780 | 22220 | 0.5854 | 0.6049 | 0.0734 | 726 | 751 | 90
0.7 | 100000 | 27000 | 99000 | 101000 | 26730 | 27270 | 0.6906 | 0.7127 | 0.0857 | 857 | 885 | 106
0.8 | 100000 | 32000 | 99000 | 101000 | 31680 | 32320 | 0.7879 | 0.8122 | 0.0753 | 978 | 1009 | 93
0.9 | 100000 | 36000 | 99000 | 101000 | 35640 | 36360 | 0.8607 | 0.8864 | 0.0486 | 1068 | 1101 | 59
1 | 100000 | 43000 | 99000 | 101000 | 42570 | 43430 | 0.9785 | 1.0062 | 0.0920 | 1214 | 1249 | 113
1.1 | 100000 | 51000 | 99000 | 101000 | 50490 | 51510 | 1.0999 | 1.1294 | 0.0936 | 1365 | 1402 | 116
1.2 | 100000 | 56000 | 99000 | 101000 | 55440 | 56560 | 1.1695 | 1.1998 | 0.0401 | 1451 | 1490 | 49
1.3 | 100000 | 68000 | 99000 | 101000 | 67320 | 68680 | 1.3198 | 1.3516 | 0.1200 | 1638 | 1678 | 148
1.4 | 100000 | 75000 | 99000 | 101000 | 74250 | 75750 | 1.3981 | 1.4305 | 0.0465 | 1735 | 1776 | 57
1.5 | 100000 | 82000 | 99000 | 101000 | 81180 | 82820 | 1.4705 | 1.5032 | 0.0400 | 1825 | 1866 | 49
1.6 | 100000 | 91000 | 99000 | 101000 | 90090 | 91910 | 1.5558 | 1.5887 | 0.0526 | 1931 | 1972 | 65
1.7 | 100000 | 110000 | 99000 | 101000 | 108900 | 111100 | 1.7121 | 1.7450 | 0.1234 | 2125 | 2166 | 153
1.8 | 100000 | 120000 | 99000 | 101000 | 118800 | 121200 | 1.7836 | 1.8163 | 0.0386 | 2213 | 2255 | 47
1.9 | 100000 | 130000 | 99000 | 101000 | 128700 | 131300 | 1.8490 | 1.8814 | 0.0326 | 2294 | 2336 | 39
2 | 100000 | 150000 | 99000 | 101000 | 148500 | 151500 | 1.9641 | 1.9958 | 0.0827 | 2437 | 2478 | 101
2.1 | 100000 | 180000 | 99000 | 101000 | 178200 | 181800 | 2.1062 | 2.1365 | 0.1104 | 2614 | 2652 | 136
2.2 | 100000 | 200000 | 99000 | 101000 | 198000 | 202000 | 2.1853 | 2.2146 | 0.0487 | 2712 | 2749 | 60
2.3 | 100000 | 220000 | 99000 | 101000 | 217800 | 222200 | 2.2545 | 2.2829 | 0.0399 | 2798 | 2834 | 49
2.4 | 100000 | 270000 | 99000 | 101000 | 267300 | 272700 | 2.3950 | 2.4211 | 0.1122 | 2972 | 3006 | 138
2.5 | 100000 | 300000 | 99000 | 101000 | 297000 | 303000 | 2.4626 | 2.4873 | 0.0415 | 3056 | 3088 | 50
2.6 | 100000 | 360000 | 99000 | 101000 | 356400 | 363600 | 2.5713 | 2.5938 | 0.0840 | 3191 | 3220 | 103
2.7 | 100000 | 470000 | 99000 | 101000 | 465300 | 474700 | 2.7114 | 2.7305 | 0.1177 | 3365 | 3390 | 145
2.8 | 100000 | 560000 | 99000 | 101000 | 554400 | 565600 | 2.7915 | 2.8084 | 0.0609 | 3464 | 3486 | 74
2.9 | 100000 | 750000 | 99000 | 101000 | 742500 | 757500 | 2.9049 | 2.9186 | 0.0964 | 3605 | 3623 | 119
3 | 100000 | 1000000 | 99000 | 101000 | 990000 | 1010000 | 2.9945 | 3.0054 | 0.0759 | 3716 | 3731 | 93
3.1 | 100000 | 1600000 | 99000 | 101000 | 1584000 | 1616000 | 3.1022 | 3.1095 | 0.0968 | 3850 | 3860 | 119
3.2 | 100000 | 3300000 | 99000 | 101000 | 3267000 | 3333000 | 3.2010 | 3.2048 | 0.0915 | 3973 | 3978 | 113
3.3 | 100000 | DNP | 99000 | 101000 | #VALUE! | #VALUE! | 3.3000 | 3.3000 | 0.0952 | 4096 | 4096 | 118

G-Pereira, 2020-12-27:
A discussion about this issue on Discord was carried through and we found a promising solution.
So a feeder needs to know in which slot it is in. Our idea was to have the feeder floor output a periodic serial signal on one pin. That serial signal would be a code that IDs the floor and can be set with normally connected solder jumpers which can be easily cut with a knife.
[image]
For this we can use a serial encoder IC like the [HT12E](https://www.farnell.com/datasheets/1899539.pdf) that has 12 pins that can be set with an address and then it outputs that code onto a pin in serial. The frequency of such signal is simply set with a resistor between two on its pins and require no more logic for the floor. The floor would be a HT12E, a resistor, solder jumpers and the connector.
The connector would be 5 pins (VCC, GND, RS-485, RS-485, ID)

PS: HT12D has reached end-of-life so a similar IC needs to be found (there are plenty of them) but this is the idea. What do you think?

vysocan, 2020-12-28:
Hello,
IMHO the DS28E05 is the best option.
The use of parasitic/phantom power from the bus is actually an advantage to
get the answer from the chip only when you expect one.  Second the size of
DS28E05 is perfect fit, and third future expandability.

G-Pereira, 2020-12-28:
@vysocan I agree with you since those serial encoders are hard to find with 8 bits and widely available on both LCSC and digikey/mouser/farnell. The HT12E used to be insanely popular but it sadly reached end-of-life.
A one-wire epromm that is flashed at the time the machine is built is a great idea. It's the same concept as mine but with programmable ID code.

daveismith, 2020-12-28:
An eeprom introduces a programming requirement as well as needing a way to visually indicate the programmed value. Not by any means insurmountable, but we would need a plan. 

I could imagine that the programming of the slot values could be done by a feeder which is moved from slot to slot. If so, this will require extra support in the rs-485 protocol and either custom support in OpenPNP or instructions (eg. How to manually send g-code to the mobo to achieve this). Another option is a completely separate fixture. The fixture option is something that makes build your own index significantly more involved, and it also complicates any production process that would happen by either Stephen or a factory. (ie, now there’s a programming step and each unit will need to labeled.) 

Part of my day job is working on production line development for projects that build tens of thousands of units a month. Programming/functional test stages are always something we try to optimize heavily because a few seconds per unit can significantly impact throughput can build cost. In addition, don’t underestimate how tricky a process applying an instance specific label can be, even for a professional contract manufacturer. In my experience, if a label is unique to a single instance (eg, an IEEe address or config property) of a product, getting accurate labels requires strict process control and also requires the ability to print labels on demand based on data read from the instance to be labeled or a database containing information about that instance. 

If the approach to building is always that the end user would program these eeproms as a part of the machine setup, you’ll still want a visual indicator for maintenance. I would suggest we introduce a blank space on the silkscreen that the slot num could be written in permeant marker or to which a label from a label maker could be applied.

Rendman, 2020-12-28:
I know that this is a contentious subject, with a lot of different idea proposed and rejected. I understand that there is a lot of ways to handle this, all would work but some with more advantages then the others.
I wanted to offer my idea, which I think has several advantages and avoids some issues.
I prioritized design goals that I gleaned were important to Stephen and the team:
1)	Cost
2)	Ease of programming/use
3)	Size
Overall, my solution is not perfect, but it achieves to some degree each item.
The voltage divider solution proposed before would be the cheapest and the simplest. However, 1% resistors would be required and MCU ADC trimming would be required to get repeatable results. It would work but could cause some issues as well.
Discussed on the Discord channel on 27 Dec was to use an encoder to serialize the state of dipswitches or solderable jumpers. The chip proposed is in a larger form factor and is hard to find.
My design is to use a cheap (ATTiny204) MCU with some dead simple firmware to essentially recreate the function of the encoder described above. I know that software has an investment cost, plus the programming issue, but I think the pain of that would be worth it.
The MCU is priced at around $.50 at quantity 1 pricing at the major suppliers (Digikey, Mouser, etc). The only other components required are the connectors (3) and a decoupling cap. The other jumpers in the circuit are solderable.
The ATTiny204 supports one wire serial connections via it’s UART. Therefore, the interface code in the feeder firmware should be simple as well. One large detractor from this idea is creation and maintenance of the firmware. However, I don’t see that it would change a lot. (I volunteer a tribute, as well)
The overall function of the MCU would be to, on power up: 
1)	Read the state of the solderable jumpers. (6 bits total = 63 positions)
2)	Convert that to a byte.
3)	Wait an amount of time.
4)	Transmit the value on the one wire serial connection
5)	Listen for an ACK from the feeder for 500 ms.
6)	If no ACK, retransmit ID, return to step 5.
7)	All done, loop forever, or enter low power standby mode.
Programming the MCU is accomplished via Microchips UPDI interface. We only need to present three pogo pads on the PCB and we can use a ~$10 programmer to program the chip in place. (Cost is less if you own an Arduino.)
I’ve attached a schematic of the proposed solution. I also have a previous rev of the PCB designed, so I know everything fits on the space already occupied by the current feeder floor.
Looking forward to all the spears… hurt me, boys.

Schematic:
https://drive.google.com/file/d/1NCuDdxIj-RdYIX4p9Ve3lshKShAJaNSG/view?usp=sharing

cad435, 2020-12-29:
To conceive things said on the discord:

When using an eeprom we can utilize the feeder software to programm the adresses in. Workflow could be something like this:
Send an RS454 command to an existing feeder that those feeder enter feeder-Floor-Programming mode. In this mode you only have to insert this feeder into one position after another. The feeder would then program every eeprom with an adress, inkrementing the adress on every power-up.
After finishing send another RS454 command to turn it back into a normal feeder

Note that for that to work we need a flash or eeprom wich can be programmed with logic-level voltage

vysocan, 2020-12-29:
@Rendman, this looks like a nice solution. However I think there is more
complexity in it over simple SOT-23 1-wire EEPROM. Not only you need to
program it, then you also need to configure the pins.
https://cz.farnell.com/w/c/polovodice-io/pameti/eeprom?typ-pametoveho-rozhrani=seriove-1-wire&druh-pouzdra-pameti=sot-23&pocet-pinu=3pinu

sphawes, 2021-01-03:
Hey all!
For posterity's sake, I'll mention the decision here as well. We're going ahead with a 1Wire EEPROM! Here is the reasoning: 

Pros:

- It's pretty cheap per part. Not as cheap as using resistors for a voltage divider, but manageable.
- Only one pin is required on the spring finger connector.
- It's expandable, so we can easily support more feeders on the same RS-485 bus, or even store other data in them if we want to.
- All feeder floor boards have the same BOM, which greatly simplifies production.
- It's reliable. Resistor voltage dividers would likely work well with the sufficiently precise resistors, but digital data storage is definitely preferable to analog.

Cons:

- The ID isn't visible inherently by observing the components. This can be mitigated by adding some silkscreen area on the board to write it!
- This method requires programming. This is acceptable only because 1) it's a one time process that only needs to happen at setup, and 2) it can be done by the feeder, so anyone can do the programming pretty easily.

I'm looking into the DS2431 due to it's relatively low cost and availability.

crazy-logic, 2021-01-03:
HI Stephen et al, 

Have you thought about using the following fact "Each DS2431 contains a unique ROM code that is 64 bits long. The first 8 bits are a 1-Wire family code. The next 48 bits are a unique serial number." this is how i'm doing ID along the bus for some 1-wire temp sensors. Using these you could search for connected devices by ROM address, then have a small app that you can use to program a Feeder ID to. (Or just use that address - not sure on how the upper software stack is dealing with IDing of feeders at this time. Certainly using something software based upstream that means you don't have to program all the feeders one-by-one would be a time saver.) if there's a spare pin on the micro after the 1-wire for the EEPROM, an ID led for IDing that doubles as an error/jam flag would also be useful.

crazy-logic, 2021-01-03:
Also - if this is based on the STM32F0 range they also have unique ID's - in the manual "33.1 Unique device ID register (96 bits)" using the unique ID's from factory and software, you could program feederID's once online and then just loose them on powerloss. 

There's also this note if BOM is becomming a problem, https://www.st.com/content/ccc/resource/technical/document/application_note/2e/d4/65/6b/87/dd/40/25/DM00049914.pdf/files/DM00049914.pdf/jcr:content/translations/en.DM00049914.pdf basically using the FLASH in the STM to store the values.

jdaandersj, 2021-01-24:
Hi all, not super familiar with the project - just watching from a distance. I spotted the video today about selecting 1-wire to solve this problem.

I would just like to point you at [PJON](https://github.com/gioblu/PJON), specifically this [implementation of PJDL](https://github.com/gioblu/PJON/tree/master/src/strategies/SoftwareBitBang) which allows single wire (+ground) serial communication. It runs on ATmega328P, ATtiny85 and others so could be worth considering. Also multi-master is supported.

The main bonus is that it's open-source and since you'd have a microcontroller in place of the 1-wire EEPROM, you could add extra functionality later if needed on that end of the connection. ATtiny85 is comparable cost to the EEPROM mentioned in case that is a concern (does also require an external crystal since the internal RC oscillator is not great).

If you have any questions, there is a [gitter chat](https://gitter.im/gioblu/PJON) where the inventor (@gioblu) is very active.

G-Pereira, 2021-02-04:
Like it was mentioned on Stephen's videos this amazing discussion have culminated on using the OneWire EEPROM. Thank you all for this amazing research work!

closed 2021-02-04 by G-Pereira