opulo-inc/lumenpnp #64: Marlin support for RS-485

sphawes, 2020-12-04:
We are using a few UART ports on the STM32F4 to communicate with feeders and other peripherals over RS-485 (thanks to the MAX3485 chip). Marlin does not currently have support for RS-485, or generally sending data out over a second serial line.

kiesel-electronics, 2020-12-21:
Did you already decide, which protocol you want to use?
If not, I would suggest Modbus in RTU mode.

- The protocol is very simple and has been around since the late 70s.
- The specification is publicly available (Modbus Serial Line Protocol and Implementation Guide V1.02).
- Several Open-Source drivers are available, so implementation might be faster.

nielsnl68, 2021-01-23:
> Did you already decide, which protocol you want to use?
> If not, I would suggest Modbus in RTU mode.

Hi, i have been looking into Modbus myself for an other project but looking at it it screams to me it have a lot of overkill when you just want to send some commands to client devices etc.

Maybe i'm not seeing the pros correctly, so could you suggestions on why the modbus protocol is the way to go within this project?

kiesel-electronics, 2021-01-26:
Hi,
the Modbus protocol is as simple, as you could get in such an application.
The documentation might be intimidating at first, because many of the documents talk about Modbus TCP/IP, which we are not interested in.
The most important document is "Modbus Serial Line Protocol and Implementation Guide V1.02".
There, you can ignore ASCII mode, it is entirely obsolete.
RTU Mode is, what is used in most applications.

[image]

A Modbus frame consists of the slave address, a function code, some data and a CRC checksum. You can’t do any less, if the protocol should be of use in a real application.
The function codes are defined in the “MODBUS Application Protocol Specification V1.1b3”. At the core, this is also quite simple, basically it allows the master to read and write single bits or 16bit words.
The wording is a bit old fashioned in some parts. For example, when they speak of “Coils”, they mean a simple bit.
All the rest isn’t really needed but can be useful for diagnostics. Experience tells that sooner or later most of these functions will be needed anyway.

closed 2023-08-30 by G-Pereira