opulo-inc/lumenpnp #65: Marlin support for ADC

sphawes, 2020-12-04:
We intended to use an ADC on the STM32F4 to read an analog signal from the vacuum sensors. This is not a feature that is currently supported by Marlin.

zero48, 2020-12-05:
Hello, I been following this project since day One .
I did a project were I needed check Pression, However I couldn't use the MCU's ADC for a specia reason.
What I did and solved my impasse. I used the sensor MPS20N0040D-D which it is inexpensive combined with  theSigma/Delta 24bit  HX711 which can be used in I2C and  SPI.  Maybe that approach may help (in my humble opinion). 
Here is a good place to get know that Pression sensor despite that the HX711 was not used and lead to misjudge. Once you transport to HX711 Everything works flawlessly.

bholmes451, 2020-12-05:
What about using the Analog Sensor functionality in Marlin which is used for Thermocouple input? Marlin does use analog sensing for the Hotends and Heated buildplate.

sphawes, 2020-12-05:
@bholmes451 True, Marlin does have support for sensing the Thermocouple input, but it doesn't seem like an arbitrary ADC pin. Looks like the code supports a peripheral chip that handles driving and reading the thermocouple. There's also some ADC usage for a filament width sensor, but also doesn't seem like it's an open ADC we can use for the vacuum sensor. The work that needs to be done here (as I understand it) is effectively creating support for a new interrupt handler for this application of the ADC. This is something that @martinbogo is working on now.

sphawes, 2020-12-05:
@zero48 it seems that the sensor you suggested seems to only support sensing positive pressure above 0 gauge psi, which won't work for our vacuum application. Thank you for the suggestion though!

bholmes451, 2020-12-06:
I guess I should have said thermistor instead of thermocouple. These pins are analog voltage sense pins (See Example Below):

`#define TEMP_0_PIN                          PC3   // Analog Input
#define TEMP_1_PIN                          PC2   // Analog Input
#define TEMP_2_PIN                          PC1   // Analog Input
#define TEMP_3_PIN                          PC0   // Analog Input

#define TEMP_BED_PIN                        PF10  // Analog Input

#define TEMP_5_PIN                          PE12  // Analog Input, Probe temp`

zero48, 2020-12-07:
> @zero48 it seems that the sensor you suggested seems to only support sensing positive pressure above 0 gauge psi, which won't work for our vacuum application. Thank you for the suggestion though!

Fair enough... XGZP101DB1R DIP package MEMS pressure sensor 100kPa pressure hydraulic suction vacuum packaging machine
Just in case same principle with HX711.

Fanfwe, 2020-12-07:
Just throwing a dumb question here, but looks like what we need to do with this sensor is to determine if the nozzle holds the part or not. The sensor is analog, so we can measure how much of a negative pressure we have, but do we actually need it ?
To me it sounds like we could simply compare the analog value with a reference threshold, and turn this into a logic signal that can go to a regular digital input of the MCU. Wouldn't that make more sense ?
What would be the use case for being able to have a variable threshold, and hence, read the analog value in software ?

kiesel-electronics, 2020-12-15:
@Fanfwe  OpenPNP has different modes for sensing parts by the applied vacuum.
1. there is an adjustable threshold
2. the pressure difference between before and after the pick is analyzed
3. the nozzle waits on the part until a certain vacuum is applied, before it picks it up.
If the ECU gets only a digital signal, the options for adaptive algorithms would be very limited

cad435, 2020-12-30:
To be correct, Marlin uses ADC for more than temperature sensing. src\module\temperature.h suggest ADC are used for:

Temperature,
Analog Joystick inputs,
Filament With sensor,
Power Current Monitoring,
Power Voltage monitoring,
ADC-Buttons

I actually didn't know that either until I looked it up.

It should be not too hard to make another class wich utilizes the ADC.

cad435, 2020-12-30:
Ah yeah, I found it. All ADC-Calls are made from inside a switch statement made in src\module\temperature.cpp around line 2704 (depending on wich version of marlin you browse).

sphawes, 2021-10-05:
This can be closed when [this PR](https://github.com/MarlinFirmware/Marlin/pull/22890) gets merged into Marlin!

stuartpittaway, 2021-11-24:
This PR is also related https://github.com/MarlinFirmware/Marlin/pull/23184

stuartpittaway, 2022-02-01:
Think this can now be closed as Marlin command is now merged.