opulo-inc/lumenpnp #69: [Mobo] IRF540N MOSFETs can not be operated directly with 3.3V MCU output

kiesel-electronics, 2020-12-15:
The IRF540N isn't a logic level MOSFET. To fully switch on, it requires about 10V at it's gate.
If driven by a 3.3V MCU output, it doesn't switch on reliably and may heat up significantly.
For reliable operation, a gate driver circuit with appropriate supply voltage is required.

g2pmas, 2020-12-18:
I use AOD4184A (very cheap on LCSC) or KIA30N06BD (LCSC PN: C134931). Often i add flyback diode, status led and a fuse.
[image]

kiesel-electronics, 2020-12-18:
Yes, those have a threshold voltage of max. 2.5V.
But still, they don't switch on fully with 3.3V. As you can see in the diagram "On-Region Characteristics" (AOD4184A), the transistor will limit the current to something between 20A and 40A.
I guess there is no use case for such high currents in this kind of PnP machine anyway, so this won't be a significant issue.
With 35um copper PCBs anything beyond 10A is also pretty impractical, because of the huge trace width required.

Probably this kind of transistor is simply overkill and much smaller ones would do fine.

sphawes, 2020-12-20:
Sorry folks, that footprint is incorrect! The mosfet I ended up getting working on this board was [this one](https://lcsc.com/product-detail/MOSFET_Wuxi-NCE-Power-Semiconductor-NCE6020AK_C108639.html). Not easily found on Digikey though. Looking at the TSM900 as a replacement. Max gate threshold voltage is 2.5v, should be able to be triggered with a 3.3V GPIO just fine. I need to update the symbol to fix the naming, if the TMS900 seems like a good option.

kiesel-electronics, 2020-12-20:
[image]
That is still a bit close for my taste, at 3V it's doing almost nothing.
I would recommend to put in some level shifting to 5V, just to be sure.
Something like https://www.mouser.de/datasheet/2/308/MC74VHC1G125_D-1387542.pdf should do fine.

kiesel-electronics, 2020-12-20:
For the TMS900, I'm not sure. The datasheet doesn't tell us RDS(on) for anything below 4.5V, so I would geuss, it is not really on below that.

gasmeter, 2020-12-21:
Hi The FET needs to be turned hard on and quite quickly.
If low speed then a small fet can be used as a level shifter or better still a proper fet driver from someone like maxim.
Something like a max4426 series.
Peter

kiesel-electronics, 2020-12-21:
@gasmeter Yes, it would be best, to use a dedicated MOSFET driver for switching.
The maxim parts are a bit expensive. Something like the FAN3100TSX should also do.
If the board should run on more than 12V, a dedicated power rail may be required.

Since we are not doing anything like PWM, switching loss shouldn't be much of an issue.
Ground lift is always an issue with power circuits. Careful design of the ground plane is required. Driving the gate with 10V would make this also more robust.

TPL7407LPWR isn't really meant as gate driver. For some odd reason, Mouser puts it in this category.
It is a nice replacement of the famous ULN2003. Toshiba makes something similar (TBD62083A).
With a pull up resistor it could be used as gate driver, but switch on wouldn't be very fast. 
Also, the signal is inverted. 
I would avoid a solution like this, because if the MCU is in reset for whatever reason, or the 3.3V rails fails, all power outputs may get switched on.

gasmeter, 2020-12-22:
Hi
This might work for you.

If the voltage rail is higher than 12V add a zener on the gate.

[image]

Use either a multi package inverter or single gate versions etc.
The 50R in series with the gate stops oscillations at rf frequencies.
Peter

kiesel-electronics, 2020-12-22:
@gasmeter The 74HC4049 requires a 5V supply. If that fails, the output clamping might pull it down anyway.

If a level shifter is used, something like this might also work:
[image]
I have this from the Duet2 schematic.
The signal from the MCU is still inverted, but that shouldn't be an issue.
But if the 5V rail get's shorted, the issue is still the same...

gasmeter, 2020-12-23:
I agree something along those lines 

I think it would make sense to generate a supply OK signal for the entire board.
Take the various supplies etc and any critical signals and produce a signal that can be used as an interlock.

Example using the interlock signal.

[image]

Peter

kiesel-electronics, 2020-12-23:
I made a little schematic, how the power good signal could be created.
The LM339 is a quad opamp with open collector outputs and a max supply voltage of 36V.
[image]
First, I create a 2.5V reference voltage. I use a zener and a resistor here, but any method would do.
Then an opamp is used to compare 3.3V against the reference. The same is done for 5V, but with a voltage divider in the front.
More voltage rails could be observed like this. The outputs are connected together and pulled up by a 10k resistor (wired AND).
The fault should be the dominant case, so positive logic must be used. 
Last, I (mis)use a opamp as inverter to generate the nPowerGood signal we want.
Because the whole circuit is driven by the rail, which is switching the MOSFET, nothing would happen, if this rail fails.

kiesel-electronics, 2020-12-26:
@gasmeter I think, the 74HC4049 wouldn't work well with 3.3V signal input (if used with 5V Vcc), because VIH (input high threshold) is already 3.15V at 4.5V Vcc.

Parts from the HCT or AHCT (the T is important) logic series have a fixed VIH at 2V. So 74HCT04 should work fine.

G-Pereira, 2020-12-26:
This issue was discussed on Discord and in order to keep BOM cost low, a MOSFET with signal-level gate characteristics was preferred and PR #78 addressed this issue

PR #78 (2020-12-29): Add MOSFET driver and freewheeling diodes
Fixes #69 and #70, Be sure to merge PR #74 first!
[image]
closed 2021-01-27 by G-Pereira