opulo-inc/lumenpnp #76: Left limit switch won't work

JudgeBeeb, 2020-12-19:
The left limit switch won't work because the x-gantry-back fouls on the rear nyloc nut holding protruding from the left y-gantry-support just before the edge of the x-gantry-front fully depresses the switch. Adding a small bump to the left edge of x-gantry-front or making the x-gantry-back slightly narrower would resolve the problem.

sphawes, 2020-12-20:
Thanks for the report @JudgeBeeb! I didn't have this problem with my machine as my bolts are only 60mm long, but it's still a very close fit. 

I like the approach of adding a small cutout on the -X side of x-gantry-back.

daveismith, 2020-12-26:
Can this be solved by simply switching the direction the bolt is inserted into the left y-gantry? Because I was using 60mm bolts which were lengthy, I put the head of the bolt on the inner side of the y-gantry and then had the nyloc nut on the outer side. This means that the switch can engage when the x-gantry-front is fully at the left.

JudgeBeeb, 2020-12-27:
This won't work for me. I am using 45mm bolts which are barely long enough to engage with the nyloc nut on the other side. It is the nut itself which stands some 4.8mm proud of the y-gantry which is causing the problem. If I switch the bolt around then I will have the same problem with the head of the bolt which is actually another 0.2mm in depth (5mm). So the situation will get worse, not better. I just need an extra millimetre to fully depress the micro-switch.

[image]

G-Pereira, 2021-02-05:
This was already fixed!

G-Pereira, 2021-02-05:
Apparently not! :)

G-Pereira, 2021-02-23:
This has been fixed on the cad-pn branch which is an ongoing effort to fix topology issues. It was created an insert for the nut and a screw with 5mm less length should be used instead. The STL can also be found there.

closed 2021-02-23 by G-Pereira