opulo-inc/lumenpnp #80: No z-gantry-right design in the 'freecad' branch

JudgeBeeb, 2020-12-21:
There is no z-gantry-right design in the 'freecad' branch. The component shown in the 'Switching the Index PnP to FreeCAD!' video is not a precise mirror image of the z-gantry-left part and is also different from the corresponding part in the 'master' branch.

G-Pereira, 2020-12-26:
There are several commits since the day that video was published (12th December) probably it was improved already!
The parts on the freecad branch are the most up to date. Some days ago a script was added to generate STL automatically to avoid having out of date STL files in the repository, so you need to download them again.
Can this issue be closed?

G-Pereira, 2020-12-26:
@JudgeBeeb Can this issue be closed?

JudgeBeeb, 2020-12-26:
This is still a live issue. The part is missing from the 'freecad' branch.

daveismith, 2020-12-27:
Is it not just a mirror of the left part? That's what I ended up printing and am planning to install. Agree it should be in the cad though so that the STLs are getting exported.

JudgeBeeb, 2020-12-27:
I too have printed a mirror of the left part. But if you look closely in the video, the part Stephen printed is not a mirror of the left part. Not sure why though.

G-Pereira, 2020-12-27:
Now I see what you mean! @sphawes has plans to revisit paste extrusion / second head later when everything else is rock solid. This is on his low priority list.
Let's keep this open to remind us to add that part!

G-Pereira, 2021-05-14:
Fixed!

closed 2021-05-14 by G-Pereira