opulo-inc/lumenpnp #86: 24V In support for the Mobo

sphawes, 2020-12-29:
Currently the board only supports 24V in. Most components are 24v tolerant, but the vacuum sensor circuitry does not. The INA126 only accepts voltages up to 18V.

This needs to be rectified somehow to allow for 24v in, and the whole board needs to be checked for 24v compatibility.

sphawes, 2020-12-29:
The INA126 also should not be powered by a switching source. This should be changed while fixing the original issue as well.

cad435, 2020-12-29:
the INA126 shouldn't be powered directly from switch-mode PSU anyway. Switching to 42V allows us to place a 12V LDOinto the rail. Given the current is rather low we can use a small SOT23-3 LDO

cad435, 2020-12-29:
I checked Branch mobo-rev02, everything is compatible with 12V except the INA126

U7 (TPS54336) is close with a max input voltage of 28V, U14 (FDS4435) has a maximum V_DS of -30V and a maximumg V_GS of +-25V (Close one!)

daveismith, 2021-01-29:
@G-Pereira, does the move to the hx711 fix this? If memory serves you were only using regulated rails.

daveismith, 2021-02-04:
So with the various proposed changes for the vacuum sensors, it seems like the INA126 will go away and that would make the board able to run from 12 or 24V. Some little things I would think worth adjusting before we have someone build one though:

- check the current limiting resistor value on the 12V led to make sure it will be a reasonable brightness for both 12V and 24V supplies
- add a selection switch / jumper to the board to control the output voltage to the feeders. rev 4+ has support for 5/12/24v supply

cad435, 2021-02-04:
@daveismith 
actually the INA126 will stay as the first High-Gain preamp. (gain of 85) as this is anamplifier specifically for those wheatstone bridges like our pressure sensor. But it would either need a 12V LDO or a reworking of resistor values to get them running of 3.3V

G-Pereira, 2021-02-04:
@cad435 according to @martinbogo which is working on it the ADC chip we chose has a PGA that should enable us to ditch the preamp. We still need to wait to see the results.

cad435, 2021-02-04:
@G-Pereira 
The INA has a gain of 85. The A2D at most an internal gain of 8.

We can't ditch the preamp, we most likely however won't need the additional Lm368

G-Pereira, 2021-02-04:
I guess I was a couple of days outdated then!

closed 2021-02-21 by G-Pereira