opulo-inc/lumenpnp #87: Ringlight PCB needs updated connector footprints

sphawes, 2020-12-29:
The ringlight PCB needs an upright 3-pin JST XH connector on back, and a right angle 3-pin JST-XH connector on the front. This will make cable routing easier for both the up and down orientations. (SMD for both.)

Also, the connector pinout should match the standard servo pinout (SIG-5V-GND), and match the mobo.

G-Pereira, 2021-01-02:
The right angle is a little harder to find but it is available [here](https://lcsc.com/product-detail/Wire-To-Board-Wire-To-Wire-Connector_BOOMELE-Boom-Precision-Elec-XH-3AWB_C141812.html) and the vertical [here](https://lcsc.com/product-detail/Wire-To-Board-Wire-To-Wire-Connector_BOOMELE-Boom-Precision-Elec-XH-3AW_C18428.html)

The right angle JST XH doesn't seem to be on KiCAD

G-Pereira, 2021-01-05:
The vertical SMD JST-XH is a Shenzhen special! JST itself doesn't have it

G-Pereira, 2021-04-19:
Added!

commit 05df9a42 (2021-04-19): Fixes #87
closed 2021-04-19 by G-Pereira