opulo-inc/lumenpnp #93: [MOBO] Change to a 4 layer board

G-Pereira, 2020-12-30:


Rendman, 2021-01-01:
This could drive up the PCB cost for those people that want to DIY the motherboard. Is having 4 layers absolutely necessary?

G-Pereira, 2021-01-01:
Nowadays 4 layer boards are close to 2 layer boards price.
The 4 layer switch will allow a much nicer ground plane. It is a mess atm

plasticuser, 2021-01-01:
Frankly, the cost saving of using a 2 layer PCB is so small compared to the investment of building the machine, it's a senseless economy. What you gain in signal isolation and EMI reduction from going 4-layer is worth ten times the cost.

This controller board is really a digital power controller switching some quite high currents, relatively speaking. This makes return paths critical to the design. Going 4-layer allows a single plane just for a large ground fill. It renders concerns about AGND and DGND irrelevant. 

There's no reason to use a 2-layer board except sadomasochism.

sphawes, 2021-01-03:
> There's no reason to use a 2-layer board except sadomasochism.
@plasticuser Ok, that's a bit much. At small quantities, the cost of this board doubles when switching from 2 to 4 layers. And even at moderate scale, it can still represent a ~50% cost increase. (Using PCBWay for quotes.) Just because it might be small in comparison to the cost of the entire machine doesn't mean it's something to disregard, especially if we haven't seen any concrete issues that are due to using 2 layers.

That being said, I still think it's worth the cost increase. Even though it is technically working as-is, some of the return paths are a bit thin for my liking and it would make routing a tremendous amount easier. As for folks that want to build it themselves, the increase is about $12 US per board. I think this is acceptable if it helps ensure they're less likely to have issues with spinup.

This is something I'll work on once the next revision of feeders are complete!

plasticuser, 2021-01-03:
I looked at the PCB, and for me it isn't really too much of a concern thatr the return paths are thin, as much as that they're indirect. A lot of various switching currents are driven through shared ground paths that do introduce a fair amount of potential cross-talk. If a layer was reserved for a ground fill, and nothing was routed on that layer, all signals would have direct return paths matching the outbound paths.

Another nicety is being able to push paths off the top layer to allow cleaner power provision and more strict debouncing - assuming power and ground weren't closely coupled (ie: layers 1/2 or 3/4)... If power and ground planes ARE closely coupled, well, hehe, debounce positioning can be quite.... relaxed ;)

closed 2021-01-28 by G-Pereira