opulo-inc/lumenpnp #97: RC-HF filtering for endstop inputs

cad435, 2020-12-31:
The endstop inputs should have a little RC circuit wich may or not may be populated to get rid of unwanted EMI noise if its occouring. 

It have to be a high filter frequency though to not disturb the Homing to much.

An RC-Filter of 100Ohm and 10nF will result in a pole frequency of roughly 160kHz wich translates to 6.25us time delay (wich will even lower practically as voltage on the MCU pin must not rise to 100% to get recognized by the MCU). that should be acceptable.

If not populated, the R can replaced with a zero-ohm resistror or a solder bridge to give maximum amout of  accuracy when homing (althoug the physical switch used will produce more error than the RC-Filter delay i'd image)

[image]

daveismith, 2020-12-31:
Is this not a essentially a debounce circuit? If so, why not do it in software? That way there is no BOM cost.

cad435, 2020-12-31:
Yes it is. but always remember that you have to integrate it into marlin...

And its not worth integrating a hole debounce-circuit into marlin when achieving the same can be done with 2 sub-ct components.

Also the software debouncing would have more jitter than the HW-Circuitry, so this would give less accuracy.

bholmes451, 2020-12-31:
I'm surprised you would think Marlin doesn't already account for debouncing cheap endstops. In addition there is a feature which can be enabled for reducing extra endstop noise.

//#define ENDSTOP_NOISE_FEATURE

gasmeter, 2020-12-31:
Best not to filter, use a latch instead, then clear after its read.

JudgeBeeb, 2020-12-31:
I don't know if this is relevant, but this seems to be the problem I am experiencing. If I run the limit switch cable through the same conduit as the stepper motor cable, the MCU seems to pick up spurious limit switch signals. I am going to order some shielded cable to see whether that improves matters.

cad435, 2020-12-31:
> Best not to filter, use a latch instead, then clear after its read.

I don't know what do you mean? 
If you latch it on trigger than it would also false trigger in noisy enviroments (only one time then, but still trigger falsely...)

cad435, 2020-12-31:
> I'm surprised you would think Marlin doesn't already account for debouncing cheap endstops. In addition there is a feature which can be enabled for reducing extra endstop noise.
> 
> //#define ENDSTOP_NOISE_FEATURE

Shame on me, I searched the whole marlin code for "debounce" but didn't notice this config setting... Actually I never noticie it while using marlin for about 5 years now -.-

G-Pereira, 2021-01-01:
@cad435 we don't need extremely high precision on homing. The cameras do that precision work possible 😉

JudgeBeeb, 2021-01-01:
This is the schematic of the switch modules being used for the end stop.
[switch.pdf](https://github.com/sphawes/index/files/5758849/switch.pdf)

JudgeBeeb, 2021-01-01:
This is what is already built in to the module that is being used as the end stop.
[image]

G-Pereira, 2021-01-01:
@JudgeBeeb you shouldn't be shorting 3.3V and GPIO with a LED

JudgeBeeb, 2021-01-01:
> @JudgeBeeb you shouldn't be shorting 3.3V and GPIO with a LED

I know. But there doesn't seem to be any other way of connecting the header on the motherboard with the header on the switch module which will result in it working as intended.

G-Pereira, 2021-01-01:
@JudgeBeeb you can mimic the module circuit
[image]

JudgeBeeb, 2021-01-01:
Sorry. My bad, I had drawn the circuit wrong. To be fair, the image from the Marlin website is itself confusing as it has Pin 4 connected to COM when, from the photo, (and from the real module, Pin 1 is clearly connected to COM. Pin 4 is VCC.
[image]

JudgeBeeb, 2021-01-01:
The feature that @bholmes451 referred to is implemented in Configuration.h thus:
```
// Enable this feature if all enabled endstop pins are interrupt-capable.'
// This will remove the need to poll the interrupt pins, saving many CPU cycles.
//#define ENDSTOP_INTERRUPTS_FEATURE

/**
* Endstop Noise Threshold
*
* Enable if your probe or endstops falsely trigger due to noise.
*
* - Higher values may affect repeatability or accuracy of some bed probes.
* - To fix noise install a 100nF ceramic capacitor in parallel with the switch.
* - This feature is not required for common micro-switches mounted on PCBs
*   based on the Makerbot design, which already have the 100nF capacitor.
*
* :[2,3,4,5,6,7]
*/
//#define ENDSTOP_NOISE_THRESHOLD 2
```

But I do have the 100nF capacitor (as per the above circuit) and I still have lots of noise. I have tried changing the size of the capacitor to something (much) larger and it doesn't help with the false triggering.

The ENDSTOP_NOISE_THRESHOLD feature is implemented in endstops.cpp with the following comments:

```
/**
  * Filtering out noise on endstops requires a delayed decision. Let's assume, due to noise,
  * that 50% of endstop signal samples are good and 50% are bad (assuming normal distribution
  * of random noise). Then the first sample has a 50% chance to be good or bad. The 2nd sample
  * also has a 50% chance to be good or bad. The chances of 2 samples both being bad becomes
  * 50% of 50%, or 25%. That was the previous implementation of Marlin endstop handling. It
  * reduces chances of bad readings in half, at the cost of 1 extra sample period, but chances
  * still exist. The only way to reduce them further is to increase the number of samples.
  * To reduce the chance to 1% (1/128th) requires 7 samples (adding 7ms of delay).
  */

```
From the little test routine I have written, two polls separated by a 1ms delay is sufficient to address the problem. So that seems to be the answer for me.

BTW, ENDSTOP_NOISE_THRESHOLD is disabled in the Configuration.h file on our repo.

daveismith, 2021-02-03:
@G-Pereira, it seems like the recommendation from @JudgeBeeb is to use the software filtering instead of hardware as it was more reliable. Should we create a new ticket asking for that, link it to this one and then close?

JudgeBeeb, 2021-02-03:
Yes.

Setting line 720 of Configuration.h to `#define ENDSTOP_NOISE_THRESHOLD 2` addresses the problem just fine.

I have also purchased some shielded cable to see if that resolves the problem, but haven't had a chance to try that yet.

cad435, 2021-02-03:
I'm always a friend of "having one more option" so we might include the bare footprints so if someone decides they want the filtering, they can populate the components...

But it should be theoretically as good in software as it would be in hardware...

JudgeBeeb, 2021-02-03:
If we are using the endstop modules, they already have an RC filter on the module (as per the above schematic) but they are insufficient to stop the false triggering. I replaced the capacitor with the biggest 0805 I have in stock (47µF) and whilst it reduced the frequency of the false triggers, it didn't eradicate them. I haven't scoped the signal yet (oscilloscope is on the other side of the lab). But I suspect that if we are running thin, unshielded, cable for both endstops and stepper motors in the same conduit we are going to get an awful lot of noise and I don't think an RC filter is ever going to be an adequate solution.

G-Pereira, 2021-02-04:
Nice to know you are getting good results with the software filtering @JudgeBeeb !@daveismith I don't think we need another ticket. This addresses the issue and includes a discussion so it has all the information regarding the change.

closed 2021-03-03 by G-Pereira