opulo-inc/lumenpnp #98: FE1.1s Alternate Part

daveismith, 2020-12-31:
The FE1.1s USB Hub is a part that is not available from electronics distributors outside of Asia. In addition, at the time of ticket filing, LCSC currently has a total stock of 14. This is rather concerning from a supply chain perspective. A suggested alternate is the [MaxLinear XR22404CG28TR-F](https://www.mouser.com/datasheet/2/146/xr22404-1856004.pdf) which is available from Mouser (which currently has about ~2600 in stock). 

The part uses the same footprint and a pinout which can be made compatible with a small number of changes which I've listed below.

Pin | FE1.1s | XR22404 | Notes
--- | ------ | --------- | -----
12 | VD18_O | NC | Add a zero ohm resistor to allow this to be NC on a board using the XR22404
13 | VD33 | NC | Add a zero ohm resistor to allow this to be NC on a board using the XR22404
18 | VBUSM | VBUS_SENSE | The XR22404 has the following note: <br>Active high. Indicates connection to USB host. Connect to VBUS power input from USB host through series 10kΩ registor[sic]. A 100kΩ pull-down registor[sic] and 100nF shunt capacitance are required on this pin.
19 | BUSJ | BUS_PWR# | On both data sheets this indicates it should be strapped high or low to configure the device as bus powered or self powered. The MOBO don't currently do that which could lead to inconsistent behaviour.
21 | VD33_O | VCC33 | The two datasheets require a different value decoupling capacitor when the device is being powered by 5V. Set up BOM variants for this with C18 = 10uF for FE1.1s and C18 = 4.7uF for XR22404
28 | VD18 | NC | Add a zero ohm resistor to allow this to be NC on a board using the XR22404

cad435, 2020-12-31:
both decoupling capacitors can be 4.7uF as bigger decoupling capacitors won't harm operation

arturo182, 2020-12-31:
Not sure if upgrading from a part that's only in stock from one Asian distributor to a part that's only available from one Western distributor is much of an upgrade.

[image]

The low LCSC stock is worrying, an alternative should be looked into.

daveismith, 2020-12-31:
@arturo182  - This suggested part is pin compatible with the noted changes. Meaning one could order either part and use it depending on availability. I went into this looking for any alternate without the constraint of being the same footprint.

The options available outside of the FE1.1s are significantly more expensive and most of them would require a completely different footprint. The cheapest options I found with a quick search tended to be from Microchip. If we were to look at changing USB hub chips, that would need to be a further discussion around the cost and layout impact.

Right now the only requirements I'm aware of for this component are:
- USB 2.0 support
- 3+ downstream ports
- minimal external components required.

This issue would be a good place to collect the possible requirements to allow a decision to be made wrt to the effort / risk tradeoff we're interested in making.

JudgeBeeb, 2021-01-04:
[Utsource](https://www.utsource.net/itm/p/11199771.html) has over 26,000 of the FE1.1s in stock at present. Albeit, with a minimum order quantity of 5. Failing that, there is the FE2.1, which is pretty straightforward to use and there is plenty of stock at [LCSC](https://lcsc.com/product-detail/USB-ICs_Terminus-Tech-FE2-1_C39693.html).

daveismith, 2021-02-21:
A possible alternate for the FE1.1 is the [MaxLinear XR22414](https://www.maxlinear.com/ds/xr22414.pdf). I haven't yet evaluated the different strapping, etc.

G-Pereira, 2021-02-21:
The only issue I find is being really expensive, but it is better than nothing, I have added as an alternative! Nice work!

closed 2021-02-21 by G-Pereira