opulo-inc/lumenpnp #99: Update R14 to Standard Value

daveismith, 2020-12-31:
R14 is currently 720R in the schematic, but this is not a standard series resistor value. Some assembly services may not offer this value for resistors. Given this is a current limiting resistor, and there are already 680R resistors on the board, it is recommended that R14 be updated to 680R.

daveismith, 2020-12-31:
I'll take care of this as part of my PR for automatic BOM & schematic generation.

G-Pereira, 2021-01-02:
Hi @daveismith the 720 Ohm resistor is for the 12V rail LED. Reducing its value will increase current. I think it would be better to change it to 1k ohm resistor instead since it is also used on the board and results in 10mA LED current, which is more than enough!

daveismith, 2021-01-28:
I'm fine with 1k. It's a value that's already in the BOM.

closed 2021-02-02 by daveismith