opulo-inc/lumenpnp #107: Soldermask issue

G-Pereira, 2021-01-05:
Soldermask clearance and solder mask minimum width should be set to zero in **EVERY** board. This improves solder quality for LQFP chips, otherwise solder bridges can be frequent.
Current soldermask:
[image]
How it should be:
[image]

wilhelmzeuschner, 2021-01-05:
What I believe to be the case is, that Stephen probably took this value from JLCPCB's Website: https://jlcpcb.com/capabilities/Capabilities
[image]

That being said, I too have always used a 0 in that setting for all of my orders (from JLCPCB) and have never experienced any issues with low-pitch packages.

bholmes451, 2021-01-05:
The other very important factor to consider is the Stencil thickness for these parts. If you are getting bridges you may need to use a thinner stencil. For a couple of my designs simply going from the default .12mm thick stencil to a .10mm stencil eliminated the bridging.

G-Pereira, 2021-01-06:
@wilhelmzeuschner that's unlikely since the set values are KiCAD defaults. KiCAD recommends setting this value to zero and the fab does the best he can.
@bholmes451 nice to know! The paste clearance and minimum width are both set to zero so this issue doesn't change the paste layer

wilhelmzeuschner, 2021-01-06:
@G-Pereira weird, when I start a new blank project these values default to 0. Maybe KiCAD remembers the last setting, although that doesn't seem to make much sense. Why would the default value be different from the recommended one anyways?

But that's not really the point, as said earlier, I completely agree with you, that this should probably be set to 0 in this case. :)

G-Pereira, 2021-01-06:
@wilhelmzeuschner That happens to me as well when I start a new project on my computer. It is always set to zero because I've changed it once to zero

commit edd6be45 (2021-01-28): Fixes #107
closed 2021-01-28 by G-Pereira