opulo-inc/lumenpnp #117: [CAD] - add a tensioning screw to the Z-Side of the Y-gantry.

cad435, 2021-01-28:
Right now the tension is only applied from one side, that will bend the lower V-slot wheel.
[image]
[Issue output v1.zip](https://github.com/sphawes/index/files/5888909/Issue.output.v1.zip)

G-Pereira, 2021-01-30:
This is rather a dirty fix cad435! :D Do you have any other suggestion to this? (most probably with changes to the Z rail)

cad435, 2021-01-30:
Actually no.
This is placed so it does not interfere with anything and the screw is aligned so.it would make contact with the bolt over the whole slot.

I use this technique often, But I'm open to other approaches!

vysocan, 2021-03-13:
Hi @cad435, when I open your file in FreeCAD, I see a bit crippled, that is with mission holes and slots for M5 nuts. Please, any idea what do I do wrong?

cad435, 2021-03-13:
Just opened FreeCAD and loaded the step into it:
[image]
Can you please make a picture how its looking for you? What exactly do You mean with "a bit crippled"?

vysocan, 2021-03-14:
I have linux FreeCAD 0.18.4:
[image]
[image]

cad435, 2021-03-14:
Ah wait, you talking about the covered holes?

Yes, that's intentional. Look at the placement of the plugs, it is where the Cylindrical diameter gets smaller. That way a 3D printer would print one solid layer and the small diameter on top of that layer. After printing you can just bore it out.

It usually comes out way cleaner than without

vysocan, 2021-03-14:
OK, thank you for clarifying. I've sliced it, to see it's truly just one layer. Just misunderstand the purpose. I thought I can drill it anyway, but also thought I can ask.
Thanks Adam.

cad435, 2021-03-14:
No problem :)

vysocan, 2021-03-17:
@cad435 I've printed it out. Wondering how to insert the 3rd spacer?

cad435, 2021-03-17:
wait, I just realized that this is the X-Gantry, not the Y-Gantry...

well, yeah than you unfortunately can't use it...

vysocan, 2021-03-18:
OK, then how about making it actually thicker? That would allow the tension screw to be hidden in the body.

vysocan, 2021-03-20:
@cad435 I wonder something like this would work, and print?
[image]
My CAD knowledge is really on beginning.

cad435, 2021-03-20:
I would make a 0 shaped lip around that tension screw instead of making the whole part thicker.

This acts like a standoff and a printed 1mm spacer can be used so the V-Slot wheel can rotate freely...

If I manage to do this real quick this afternoon I'll post it

vysocan, 2021-03-20:
I did not make it thicker actually. I just used the original part, and applied your tension mechanism.

cad435, 2021-03-20:
[image]
What do you think about that? The orange part is a seperate printed part...

@G-Pereira is this worked on or still open?

G-Pereira, 2021-03-21:
@cad435 Still open, we don't have anyone looking at it yet because we are yet to finish fixing the CAD

vysocan, 2021-03-21:
@cad435 Looks fine to me. Is the part still thinner like your original one?
I've tried to finish my design as well, but having problem talk FreeCad into correctly create nut pocket :(.

cad435, 2021-03-21:
errm, yes this is thiner, because I have an lasercut acrylic part laying on top of it^^

vysocan, 2021-03-23:
I see :)

G-Pereira, 2021-05-14:
Included in the upcoming CAD revision! Thank you!

closed 2021-05-14 by G-Pereira

vysocan, 2021-05-14:
Hello @G-Pereira, forgot to mention, I come up with following:
[image]
It works quite well.
Should I try to create PR or you have already some own idea.

G-Pereira, 2021-05-16:
Hello! Check the latest video! @sphawes has already solved this issue!