opulo-inc/lumenpnp #118: [Mobo] Add LEDs for Mosfet Outputs

wilhelmzeuschner, 2021-01-28:
This has already been discussed in some other issues but it seems that it hasn't been added (yet).
It would we awesome to have the ability to see the output status immediately when debugging.

This has been referenced in #69 but it wasn't what the issue was about.
And @G-Pereira mentioned it in [PR78](https://github.com/sphawes/index/pull/78)

martinbogo, 2021-01-28:
I'm all for this -- make sure the LED's don't draw significant current when
connected, and that the LED is in parallel so that you can depop the LED (
or it's resistor ) to disable the feature.

wilhelmzeuschner, 2021-01-28:
@martinbogo or even add a small solder jumper that is NO, making enabling / disabling it faster and cleaner :)

G-Pereira, 2021-03-03:
Added!

closed 2021-03-03 by G-Pereira