opulo-inc/lumenpnp #122: [Mobo] Quality of life: move position of Reset and BOOT0 buttons

wilhelmzeuschner, 2021-01-29:
In order to make debugging / programming the STM32F4 easier it would be nice if the location of the two buttons (Reset & BOOT0) could be moved away from the middle of the board.
It would be much easier to access them, if they were located on the side of the PCB.
Maybe we could even use side actuated pins, so that connectors sticking up from the board don't restrict access. Also this would make blindly pressing reset (and boot0) easier when the board is mounted far away and the buttons aren't visible.

martinbogo, 2021-01-29:
Wilhelm,

Amongst the way this is dealt with in the industry ( other than moving the
buttons to the edge ) is adding a jumper pin header to allow the buttons to
be placed offboard.   There's nothing like a big, friendly, and easy to
press button placed wherever is convenient on a machine to make things work
better.  This is how the reset buttons work on things like the external
displays on 3D printers like the Creality Ender series.

-Martin

G-Pereira, 2021-01-29:
The question is: Will we ever have some sort of UI on the machine itself?

wilhelmzeuschner, 2021-01-29:
That's a good idea, since there is quite a lot of space on the mobo, the additional headers can be added in parallel to the existing buttons. But nonetheless I'd still like to have easily accessible buttons right there on the board itself.

G-Pereira, 2021-01-30:
Actually pressing the button and going through the PC to flash it at the same time is annoying. We can save on cost and board space to have just the headers so we can use jumpers or external panel buttons. I will change the title slightly and make sure this is added to mobo-rev03. Thanks, nice work!

wilhelmzeuschner, 2021-01-30:
For another project I needed this kind of flexibly as well. I opted to overlay the footprints of a 2.0mm connector and a side actuated button. The pitch of 2.0mm is smaller than the current connectors, so it would make more sense to use a 2.54mm pitch connector instead for this project. This would also have the benefit that the clearance between the pads would be slightly increased.

2.0mm with button footprint:
[image]

I used these parts:
[Button](https://lcsc.com/product-detail/Tactile-Switches_Yuandi-TS-1037-A10B3-D2_C499330.html)
[Connector](https://lcsc.com/product-detail/Wire-To-Board-Wire-To-Wire-Connector_BOOMELE-Boom-Precision-Elec-C64658_C64658.html)

Here is how it looks (didn't solder the parts on):
[image]
[image]
[image]

G-Pereira, 2021-02-15:
This has been added to Index Mobo rev03!

closed 2021-02-15 by G-Pereira