opulo-inc/lumenpnp #125: Add Raspberry Pi Compute Module to Index Mobo

G-Pereira, 2021-01-31:


daveismith, 2021-01-31:
@G-Pereira can you elaborate on what interfaces you would want to break out on the mobo? A description of what you want to use the Pi for would be good. 

I like the idea, but think it deserves some testing. The compute power of the Pi is decent but with the computer vision pipelines in OpenPNP we should test before trying to implement this ticket.

martinbogo, 2021-01-31:
Wait, why?   That's a lot of complexity for very little actual benefit.
 You can achieve the same thing by purchasing a Pi 4, and hooking the USB
port to the mobo.

G-Pereira, 2021-02-02:
@martinbogo and @daveismith after thinking it through it makes more sense to look at this as a computer choice task.
I just ordered a Jetson Nano 2GB to test. It seems some have had great results with it since it has an AI accelerator chip, although those results are two years old. I am interested how openpnp runs on it at the moment.

sphawes, 2021-09-20:
@G-Pereira Going to close this ticket. Although there very well might be some rad little comp that could run the machine, it's more of an exercise for the OpenPnP folks to determine it, than for us to. If you find one though, we can add it to the docs as a recommended embedded choice!

closed 2021-09-20 by sphawes

G-Pereira, 2021-09-20:
Makes sense to me.

Sounds like someone is preparing something 😉