opulo-inc/lumenpnp #131: [Feeder] Crystal & Loading Cap Selection

daveismith, 2021-02-02:
The crystal and loading cap selection needs to be reviewed on the schematic / BOM.

cad435, 2021-02-03:
Oh, I can help here^^

The IC on there is an STM32F103. From what I learned today on the discord they could be given the absolute cheapest crystals and would be fine. 

This document by ST descripes the calculation in section 3.4
https://www.st.com/resource/en/application_note/cd00221665-oscillator-design-guide-for-stm8afals-stm32-mcus-and-mpus-stmicroelectronics.pdf

I'm doing the math for this crystal:
 https://lcsc.com/product-detail/SMD-Crystal-Resonators_HELE-Harmony-Elec-X3S008000BK1H-U_C254258.html

g_mcrit = 4 x 400 x (2xpi x 8 x 10^6)² x (2x10^(-12) x 20x10^(-12))² = 1.956ma/mV

This is done assuming 20pF Load capacitors

Table 6 of the Document provided above provides the information, that the SMT32F103 Series has a maximal drive capability of 5mA/V which means we are safe.

Please note that this crystal cannot be used for the mobo!

However we can also use the 8Mhz Quarz from the mobo for the feeder, as it has a g_mcrit of 0.792mA/V and is therefore even better. This quarz however needs 12pF Loadcaps. 
I suggest making a note in the KiCAD document which denotes thos two options: 
 - the same quarz as from the mobo if one want to get its first set of feeder along with the mobo, to reduce part count
 - the cheaper quarz I linked above, which can be used if one want to produces more feeder

G-Pereira, 2021-02-03:
@cad435 great work! This kind of component selection really needs to happen across the entire set of the Index boards. This is a subject for an all different issue that has low priority for now since we are still adding a lot of new features and testing our designs.
For the Mobo you already double checked the crystal load capacitance and driving capability is ok so we can use exactly the same crystal and loading caps for the feeders.
@cad435 I've just checked and the mobo's crystal ([C518155](https://lcsc.com/product-detail/SMD-Crystal-Resonators_Suzhou-Liming-Elec-3225-8-00-12-10-10-A_C518155.html)) is cheaper than the one you linked. So we have no reason not to choose mobo's crystal.

cad435, 2021-02-03:
Perfectly fine! 

I also have the Formula baked into an excel sheet  - finished it the moment you posted :D
[XTAL_Calc.xlsx](https://github.com/sphawes/index/files/5922099/XTAL_Calc.xlsx)

G-Pereira, 2021-03-08:
Ok, so this has been fixed!

closed 2021-03-08 by G-Pereira