opulo-inc/lumenpnp #135: Enhance Power distribution on the Mobo for more flexibel usage

cad435, 2021-02-03:
Currently the incoming 5V from USB is not connected to anything. That means when the Mobo is only powered by USB, the MCU can't be operated. 

In normal usage, this is not a problem. However with this setup, you can't make usage of Marlins PSU_CONTROL feature, which would become handy when dealing in between PnP-Jobs. Inbetween PnP-Jobs, there is no need to have the machine running. To simplify the process of getting going, the PSU_CONTROLL Feature would come in handy, as Marlin would turn on A predefined pin from an AUX-Header or a Spare MOSFET which therefore can turn on the AC Input of the 12/24V power supply. 
[image]

This is not possible with the current power setup, as turning off the power, power to the MCU will also be cut.
At a cost of two schottky diodes that feature could be enabled. (see picture above)

A schottky diode will effectively drop the 5V at about 300mV which leaves 4.7V USB-Voltage - thats inside the spec of 4.5V minimum USB Voltage

[image]

It can be further discussed if power to the stacked USB-A Connector should be better connected to the same 5V-Line than the Fe1.1s

cad435, 2021-02-03:
It should be further noticed, that If we want to be absolute safe we could replace D1(and possibly D2) with a ideal Diode IC (like a LTC4376). But that would increase the BOM cost more thatn just two schottkys.

daveismith, 2021-02-03:
@cad435 - as mentioned in, I've done similar on other boards I've designed. Those boards have always been such that the USB alone could power all of the components whereas here we create two classes of 5V power (ie, always on and only on with external power).

I can see the utility for what you're wanting to do but would want to make sure we consider the following:

- clear documentation that we make no claims or guarantees about any safety from this setup and it's for convenience rather than safety.
- a review of the impact of switching off the 5V to downstream USB devices while powering the hub
- the impact on the FE1.1s circuit as it talks about the capability to re-enumerate when switching between self and usb powered modes
- if this would also be compatible with the alternate usb hub from #98

cad435, 2021-02-03:
@daveismith 

1) we don't make any guarantees anyways, otherwise we have to certify the Mobo (in EU its CE certification, not sure how USA/China is handeling Certification)

2) True, but please consider that this is the same matter with the shematic now. Our FE1.1s is only connected with Data lines and GND when the mobo is plugged in with USB Only. The proper solution would be to have a USB Voltage and Current monitor and distributing the VBUS voltage to the FE and also to the Stacked USB-A Connector

3) FE1.1s can be hardwired in the same mode, as now. Stephen chose to completely ommit the VBUS-Monitoring which does come with the FE1.1s VBUSM-Pin, so its already "as worse as it can get" (Yes I know, that is a very poor argument, but thats how it is. Proper voltage and current handling could be implemented only by increasing the board components)

4) thats true, but most certainly possible, due to the fact that we are using the FE1.1s (and any possible alternate part at the absolut minimum function, no voltage monitoring, no current monitoring)

daveismith, 2021-02-03:
Some comments:
2. For the current setup if the mobo is plugged in by USB only the FE1.1s upstream port has this issue but the downstream devices won't be powered or have anything driving d+/d- since the FE1.1s won't be powered. 

In terms of the switch from self to bus powered, would that cause the re-enumeration of downstream devices? If so, that would mean turning the psu on and off would make the serial connection to the stm be re-enumerated and might break the connection (or am I misunderstanding)?

cad435, 2021-02-03:
2) true! That is absolutely true, but it only shifts the problem one step back tbh... Also, I remeber, there are a ton of USB Current supervicer IC's out there, which only are switching off the 5V rail to USB devices. I can't prove this, but due to the sheer availability of those chips i can't imagine anything bad would happen to the devices...

for the switch, I just looked into the datasheet and on page 2 it says "Automatic re-enumeration when hub switches
from self-powered mode to bus-powered mode"

So, Yes, it would auto-enumerate the devices, but I honestly don't know if the MCU with marlin would be reset by that... I can test that when I got the mobo working.

But that can be solved when hard-wiring the chip to the (more konservative) self-powering state while only monitoring the USB-Voltage which goes to the Fe1.1s and the stacked USB-A (see picture)

If you want absolutely safe you can also pack an USB overcurrent-protection IC in there as well

[image]

G-Pereira, 2021-02-03:
> It can be further discussed if power to the stacked USB-A Connector should be better connected to the same 5V-Line than the Fe1.1s

I would like to point out that according to [Wikipedia's page on USB specification](https://en.wikipedia.org/wiki/USB) USB 2.0 is only capable of delivering 500mA. I know that some computers allow a bit more but that's out of spec. Looking at the camera's specifications each one draws 150mA, which is in total is 300mA. So 200mA is left whichever peripherals we want to power on from USB if we choose to connect the cameras.
Of course this is for this specific cameras, if we replace them with beefier ones, that is problematic.

cad435, 2021-02-03:
OK, I specifically looked it up now.

If I understand that correctly, USB spec says the, impedance on Datalines should be 90R. that means the worst thing which can happen at 3.3V D+ and -3.3V D- (Data lines have 3.3V differential voltage for USB 1.1, 400mV for USB 2.0) is 7.3mA current draw.

Given that there is at least one 15k Pull up resistor on one Dataline (must be to indicate speed) which is located on the DEVICE, we will have less current draw. I think that this will never happen, as there are specific IC's out there which will ONLY shut the VBus down when there is too much current detected.
I never heard of a problem when using them. (For example the TPS2065D)

cad435, 2021-02-04:
OK, I think found the missing piece:

Section 7.1.6.1 of the USB spec sais, that reciever input WITHOUT termination must be at least 300k!
The termination resistors must be chosen, so that the resulting impedance must be 90R. 

That means, we are perfectly safe when having D- and D+ connected to an unpowered USB slave, as the only thing that can happen are about 7mA at most through the termination resistors. And those termination resistors must be spec'ed to handle that current anyway for normal USB operation and when the USB-Datalines are IDLE.

That solves two issues:
1) we know that the FE1.1s and every other HUB IC will be fine when only connected by its data lines
2) if we go the route and make the Power distribution as I propose, we can supply the stacked USB-A with the Buck converter and be sure that the USB devices attached to that connector wont be suffering when the Buck converter is off and datalins are connected. The remaining current draw from MCU and the HUB IC would not exceed 500mA through the USB connection.

I therefore would prefer to implement the power distribution from the picture of my initial posting with the diodes as schottky diodes.

daveismith, 2021-03-12:
@cad435 - With the updated USB hub design, this needs to be revisited. @G-Pereira would like to make sure that the design in the mobo-rev03 branch is correct. Please review and comment.

 The part selected is the FE1.1 ([datasheet](https://elinux.org/images/e/ee/FE11-datasheet.pdf)). I believe that the XR22414 ([datasheet](https://www.maxlinear.com/ds/xr22414.pdf)) is the appropriate alternate.

I've found the following which shows a bus-powered design of a hub using the FE1.1 along with individual port power control. That seems like more than we want to approach, but it shows that we will need to rework circuit around at least pins VBUSM and BUS_B in our design. 

Changes:
- BUS_B/BUS_PWR#/36 there needs to be a weak pulldown to GND with a strong pullup that is powered only when there is external power
- VBUSM/VBUS_SENSE/35 needs to be connected to the VBUS pin so that the device can sense when USB power is connected
- +5V net should be driven by both VBUS and the output of U12 with diodes in both places to prevent back powering.
- Insert a PTC fuse between the +5V supply and the USB downstream ports to ensure that they don't draw too much power.
- OVCB1 should be connected to the output of the PTC fuse to detect if an over-current situation has been detected
- DIS_REG should be pulled up to +disable internal regulator
- VD33_OUT should be NC
- VDD5 should not be connected (maybe put an unpopulated zero ohm in case)
- VD3_IN should be connected to +3.3V
- /XRSTJ should be pulled up to +3.3V

G-Pereira, 2021-04-05:
Added to rev03 as an optional feature configurable by solder jumpers!

closed 2021-04-05 by G-Pereira