opulo-inc/lumenpnp #151: [CAD] - non-standard axis orientation

meutzitzu, 2021-02-09:
I'm new to this project, but I have been using FreeCAD for a while. 

first thing I noticed when I started to take a look at the files is that it really seems that the entire project is using odd orientations for most of the parts. It is considered best practice to keep the Z axis up, and negative -Y forward (as it's written on the navigation cube). Is there a reason for not using this orientation scheme ? is it because that's how the STEP files were initially exported from F360 ?

this obviously isn't exactly the end of the world, but it can turn out to be quite an annoyance for people who are using turntable camera navigation, and it's just not right.

tobiasfalk, 2021-02-09:
I notized it also wehen I locked in to the Projekt. It ist becaus they Designed the Modells so that the do not habe to move the para when printing

meutzitzu, 2021-02-10:
is it because you export to .OBJ ? I know OBJ has vertical Y axis. If you export as .STL you should still have Z axis vertical

like, when I opened the individual files they still have the Y axis vertical. So idk what's the deal

mister-alien, 2021-03-08:
Seems like @tobiasfalk is implying that the models are created so that they are easy to export straight to STL & slice for 3D printing without reorienting in the Slicing program. I would generally consider your "standard" axes' orientation to be what I'm used to as well @meutzitzu , though I have not as much experience with a FreeCAD workflow vs NX.

G-Pereira, 2021-05-14:
Hey, we are using a datum plane now to define the printing orientation and based on that our CI script will export the STL in that orientation. Thank you for your suggestions!

closed 2021-05-14 by G-Pereira