opulo-inc/lumenpnp #152: [CAD] - Y endstop not hitting the gantry properly

wilhelmzeuschner, 2021-02-09:
# Description
The Y endstop switch does not hit the gantry (FDM-0005-02_y-gantry) very well. To fix this, I glued a small piece of plastic onto it.
To fix this issue, the limit switch would need to be moved further "into / (to the right on the images, see below)" the front left leg (FDM-0001-01_front-left-leg), so that the switch makes better contact with the gantry . An alternative could be to design part of the gantry differently, although this could make the part harder to print, depending on the changes. Maybe a special, larger belt clamp (FDM-0025-00_belt-clamp_Body_001) could be printed but this would not be ideal either, since this part may need adjusting from time to time and would thus change the exact coordinates when homing. Also this would introduce yet another special part.

## Here are some photos:

**Illustration of the issue:**
Green: makeshift extension
Red: edge of the printed part
Yellow: limit switch contact area
[image]

**Same image, but unedited**
[image]

**Other angles:**
[image]
[image]

JudgeBeeb, 2021-02-09:
This is already a known issue: https://github.com/sphawes/index/issues/76

G-Pereira, 2021-02-23:
Hey! Looks like you are misusing the belt clamps. Look here:
[image]
[image]

closed 2021-02-23 by G-Pereira

wilhelmzeuschner, 2021-02-23:
The problem was my use of the older design version of the clamps, which were round and also the fact, that the limit switch was angled slightly towards the gantry. This way, it never hit either of the parts reliably.