opulo-inc/lumenpnp #153: [CAD] - Upgrade to Assembly4 ?

meutzitzu, 2021-02-10:
Why use A2Plus over Assembly4 ? 

Assembly4 is more useful specifically for moving assemblies and 3D Printers/CNC machines etc. 

is there a reason for still using A2Plus ? I know Assembly4 is newer, but it's _the_ assembly workbench for FreeCAD as of now. It's incredibly simple lightweight and reliable, and fully parametric.

If there's no reason other than "ain't nobody got time fo that" for switching to Assembly4 would y'all like it if I start to switch the assembly file over to Asm4 and make a PR ?

tobiasfalk, 2021-02-10:
As you may see in #133 I already succeeded to move to A4 but if I understood it right they want to fix some problems first

G-Pereira, 2021-02-11:
Thanks @tobiasfalk that's it. @meutzitzu we are in the process of fixing broken design constraints because we are all new to FreeCAD and haven't used FreeCAD best practices from the start! :smile: 
After that's finished we are moving to Assembly4 since it would be nicer for our assembly guide. Thank you for pointing it out and I will get back to you once those fixes are done if you want to help us move to A4. Don't know if you are already on our discord (check the readme), it is easier if we talk there!

meutzitzu, 2021-02-11:
when using Asm4 it would be possible to simulate movement, collisions, etc
really really powerful 

also, has better support for fasteners than A2Plus

speaking of which, is there a reason for not including fasteners in this project ? or just the previous "have't gotten time to do that yet"

meutzitzu, 2021-02-11:
whoops, my bad 

ok, I'll go to the discord

G-Pereira, 2021-07-04:
The CAD revision has finished and merged. Now we are in a good point to move to A4. Are you interested in submitting a PR for that?

G-Pereira, 2021-07-05:
Can you comment on this @meutzitzu ?
https://github.com/index-machines/index/blob/assembly4/pnp/cad/Assemblies/left-side.FCStd

sphawes, 2022-05-11:
We'll stick with A2+ for now, closing.

closed 2022-05-11 by sphawes