opulo-inc/lumenpnp #157: FE1.1s can't provide both camera streams to OpenPnP at the same time

G-Pereira, 2021-02-15:
The recreate the issue:
- Connet two ELP cameras directly to the host and configure both on OpenPnP
- Verify that both work by using OpenPnP's camera preview.
- Disconect cameras from host and connect Index Mobo
- Connect cameras to the Index Mobo
- Restart OpenPnP
- Reconfigure cameras (due to re-enumeration)
- Check camera previews (only one of the cameras work)

A MTT capable USB hub must be found to allow multiple concurrent streams

JudgeBeeb, 2021-02-15:
Not a problem for me.

G-Pereira, 2021-02-15:
@JudgeBeeb please provide the following information of your setup:
- Operative system
- OpenPnP version

JudgeBeeb, 2021-02-15:
macOS Catalina
OpenPNP version: 2021-01-05_05-21-56.8f290c2

G-Pereira, 2021-02-15:
Thank you. I just loaded your OpenPnP machine file and the issue remains.
One camera stream doesn't work and as soon as I disconnect the working camera the other one starts working as soon as I click the "Test" button

G-Pereira, 2021-02-15:
My setup is:
- OS: Fedora 33
- OpenPnP: 2021-02-04_06-49-15.d539596

G-Pereira, 2021-02-15:
Here's a dump of my dmesg output:
```
[  340.489612] usb 1-1: new high-speed USB device number 7 using xhci_hcd
[  340.616018] usb 1-1: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.11
[  340.616023] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[  340.616027] usb 1-1: Product: USB 2.0 Hub
[  340.617638] hub 1-1:1.0: USB hub found
[  340.617686] hub 1-1:1.0: 4 ports detected
[  341.004712] usb 1-1.2: new high-speed USB device number 8 using xhci_hcd
[  341.273748] usb 1-1.2: New USB device found, idVendor=32e4, idProduct=9230, bcdDevice= 1.00
[  341.273754] usb 1-1.2: New USB device strings: Mfr=2, Product=1, SerialNumber=0
[  341.273757] usb 1-1.2: Product: HD USB Camera
[  341.273760] usb 1-1.2: Manufacturer: HD USB Camera
[  341.283261] uvcvideo: Found UVC 1.00 device HD USB Camera (32e4:9230)
[  341.337343] input: HD USB Camera: HD USB Camera as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.2/1-1.2:1.0/input/input40
[  341.516624] usb 1-1.3: new high-speed USB device number 9 using xhci_hcd
[  341.785819] usb 1-1.3: New USB device found, idVendor=32e4, idProduct=9230, bcdDevice= 1.00
[  341.785826] usb 1-1.3: New USB device strings: Mfr=2, Product=1, SerialNumber=0
[  341.785829] usb 1-1.3: Product: HD USB Camera
[  341.785831] usb 1-1.3: Manufacturer: HD USB Camera
[  341.793295] uvcvideo: Found UVC 1.00 device HD USB Camera (32e4:9230)
[  341.847631] input: HD USB Camera: HD USB Camera as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.3/1-1.3:1.0/input/input41
[  370.124513] uvcvideo: Failed to query (GET_DEF) UVC control 1 on unit 2: -110 (exp. 2).
[  452.898305] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 2: -110 (exp. 2).
[  453.066247] usb 1-1.3: Not enough bandwidth for new device state.
[  453.066253] usb 1-1.3: Not enough bandwidth for altsetting 6
[  478.439241] usb 1-1.3: Not enough bandwidth for new device state.
[  478.439247] usb 1-1.3: Not enough bandwidth for altsetting 6
[  524.744588] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 2: -110 (exp. 2).
[  524.927901] usb 1-1.3: Not enough bandwidth for new device state.
[  524.927908] usb 1-1.3: Not enough bandwidth for altsetting 6
[  572.044670] usb 1-1.3: Not enough bandwidth for new device state.
[  572.044675] usb 1-1.3: Not enough bandwidth for altsetting 6
[  615.855706] usb 1-1.2: USB disconnect, device number 8
[  615.856873] uvcvideo: Failed to resubmit video URB (-19).
[  615.860911] uvcvideo: Failed to resubmit video URB (-19).
[  615.864877] uvcvideo: Failed to resubmit video URB (-19).
[  637.902586] usb 1-1.2: new high-speed USB device number 10 using xhci_hcd
[  638.171452] usb 1-1.2: New USB device found, idVendor=32e4, idProduct=9230, bcdDevice= 1.00
[  638.171458] usb 1-1.2: New USB device strings: Mfr=2, Product=1, SerialNumber=0
[  638.171461] usb 1-1.2: Product: HD USB Camera
[  638.171464] usb 1-1.2: Manufacturer: HD USB Camera
[  638.179177] uvcvideo: Found UVC 1.00 device HD USB Camera (32e4:9230)
[  638.233000] input: HD USB Camera: HD USB Camera as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.2/1-1.2:1.0/input/input42
[  690.825280] usb 1-1.3: Not enough bandwidth for new device state.
[  690.825284] usb 1-1.3: Not enough bandwidth for altsetting 6
```

G-Pereira, 2021-02-15:
Also I am using a 3.0 USB port on the host (all my ports are 3.0).

wilhelmzeuschner, 2021-02-15:
I have exactly the same issue on Windows 10 as @G-Pereira.

G-Pereira, 2021-02-21:
The hub was changed to the FE1.1 (instead of FE1.1s) which has a MTT (with a separate TT for each port). #98 continues to be a concern but this issue is solved.

closed 2021-02-21 by G-Pereira