opulo-inc/lumenpnp #159: [MOBO] - Add more (mosfet) outputs

wilhelmzeuschner, 2021-02-16:
Currently there are only 4 high current mosfet-switched outputs. 
On my config two are used for the two LED lights (not ringlight, just 12V LED strip) and at least another one if not two is taken up by the vacuum pump(s).
This would leave no free outputs for other expansions such as solenoids etc.
Lower current ports would probably be fine as well and would need a physically smaller mosfet.
Something along the lines of: [AP2302B](https://lcsc.com/product-detail/MOSFET_ALLPOWER-ShenZhen-Quan-Li-Semiconductor-AP2302B_C406812.html)

This is more of a quality of life request but it might come in handy to quite a few people (myself included :) )

stuartpittaway, 2021-02-21:
Good idea, I use the Alpha & Omega Semicon AO3400A ([C20917](https://lcsc.com/product-detail/MOSFET_Alpha-Omega-Semicon-AO3400A_C20917.html)) parts in a lot of my project, good price and high spec.

N-Channel 30V 5.7A 1.5V @ 250uA 26.5mΩ @ 5.7A,10V 1.4W SOT-23-3L MOSFET RoHS

G-Pereira, 2021-02-21:
Nice work! I've increased the MOSFET number to 8. We just saved 1.04$ per board by using AO3400 instead!

closed 2021-02-21 by G-Pereira

stuartpittaway, 2021-02-21:
Well done