opulo-inc/lumenpnp #163: Automatic Firmware Flashing?

ogaskell, 2021-02-21:
I was just wondering how feasible it would be to setup an automatic firmware flasher on the Index? Maybe another head could be added with pogo pins, that could hook up to ICSP connections on test pads on a PCB. 

I imagine it'd only work for certain types of MCUs, and you'd have to pick which beforehand but I think it could be really useful for fully automating board production.

daveismith, 2021-02-21:
I've been thinking a bit about this as well. I'm used to bed of nails fixtures from work and my initial thought it more to put those into the fixture and have the PCB moved to that fixture and then somehow secured. When I tried this a few years ago I had relative success with a servo driven arm that held the PCB down against the fixture. 

I've also been wondering about a vacuum to keep the PCB in place. The pick head could push the board down against the fixture and then a vacuum pile be engaged to hold it in place. 

The bug benefit I see to a fixture based system instead of a head based system is the ability to parallelize. It also lets you do customized tests on a PCB which is quite important to ship reliable products in my experience.

sphawes, 2023-05-15:
Sorry for dredging up an old thread, clearing out old issues!

This is a rad idea, but doesn't fit as an issue. I'll close this, but thanks for the thoughts!

closed 2023-05-15 by sphawes