opulo-inc/lumenpnp #164: MOBO - Add alternative footprint for vacuum sensor

wilhelmzeuschner, 2021-02-21:
Currently the [2SMPP03](https://omronfs.omron.com/en_US/ecb/products/pdf/en-2smpp-03.pdf) is used:
[image]
[image]
[image]

It has footprint a that differs from other cheaper vacuum sensors available. 

For higher flexibility in sourcing parts I suggest to add a footprint for a sensor with this footprint:
[image]
Pinout:
[image]
Here are two examples for this type of sensor:
[XGZP Pressure Sensor series](http://m.eleparts.co.kr/data/goods_old/data/XGZP_SOP6.pdf) 
[image]
or [MPS3117-006GC-A2](https://datasheet.lcsc.com/szlcsc/Metrodyne-Microsystem-MPS-3118-015GC_C114719.pdf) <- Note that the last sensor is not for vacuum sensing applications! But the footprint / pinout is the same. This type of sensor seems to be very common.

I bought the XGZP sensor. Here is an [overview ](https://www.sgbotic.com/products/datasheets/sensors/02976-datasheet.pdf)of the different models: 
[image]
Here's a [PCB](https://www.aliexpress.com/item/32923552711.html)  that should have a -40kPa sensor.
Supposedly -100kPa: [XGZP101SB1 ](https://www.aliexpress.com/item/32709990616.html) 

AliExpress certainly is not a reliable source to get parts. But the flexibility to have this common type of sensor-footprint on the PCB would come in handy. The two footprints could be superimposed, the alternative is on the top layer and the current sensor is on the bottom layer, facing up through the hole.

G-Pereira, 2021-02-21:
Nice work! This makes a lot of sense, we indeed have the possibility to add both footprints.

G-Pereira, 2021-03-05:
Hey again! I just checked this in more detail and saw that LCSC doesn't have any vacuum sensors. This footprint would just fit the AliExpress sensors which we will not be able to support and thus the gain of the amplifier should be set by the user as well as the current source parameters. If one goes with Mouser or Digikey they should get our recommended sensor, the 2SMPP03.
Anyhow, I've added this in!

closed 2021-03-05 by G-Pereira