opulo-inc/lumenpnp #199: [CAD] - Include the font for the part number in the projekt files

tobiasfalk, 2021-02-28:
The path to the font for the part number in the "PN" feature isan fixed path to "/usr/share/fonts/dejavu-sans-fonts/DejaVuSans-Bold.ttf" why not include the .ttf file in the prijekt files an make an relative path like "../../fonts/dejavu-sans-fonts/DejaVuSans-Bold.ttf" or so.

G-Pereira, 2021-02-28:
This is an excellent idea since, like you pointed out on Discord, it would suit any operative system.

My only concern is if the font license allows us to keep a copy on our repo. Also, we have to find a clear way to state the author of the font as not to claim it as our own.

daveismith, 2021-02-28:
I think it should be ok. The license is https://www.fontsquirrel.com/license/dejavu-sans.

G-Pereira, 2021-02-28:
@daveismith I think so too. They say:

> The above copyright and trademark notices and this permission notice shall
> be included in all copies of one or more of the Font Software typefaces.

So we should include that in our README or Wiki somewhere. @sphawes ideas?

G-Pereira, 2021-03-04:
It has been included in the /lib/fonts folder so parts should use **../../../../lib/fonts/DejaVuSans-Bold.ttf**

closed 2021-03-04 by G-Pereira