opulo-inc/lumenpnp #209: [MOBO] - adding I2C address pads

austinscreations, 2021-03-04:
I had mentioned this on discord, and was suggested to post here. (I'm I'm doing it correctly)
With I2c being used on the project I'd imagine there might be an external port being added where people could add their own accessories, if this is the case (i haven't looked as I'm new to the project) it might be worth ensuring that ic's that are added have their address pins broken out with solder bridge pads so that the address could be changed in the event of conflicts.

NightSkySK, 2021-03-05:
I would suggest to make it in [Adafruit STEMMA](https://learn.adafruit.com/introducing-adafruit-stemma-qt/what-is-stemma) specification, which is compatible also with Grove. In this case, we should use 4 Pin JST PH 2.0mm pitch connectors footprint
[STEMMA Technical Spec](https://learn.adafruit.com/introducing-adafruit-stemma-qt/technical-specs)

G-Pereira, 2021-03-05:
@NightSkySK there will always be a handful of different connectors used by vendors. For that reason we expose i2c pin on a AUX header for general use

daveismith, 2021-03-05:
So according to [adafruit](https://learn.adafruit.com/introducing-adafruit-stemma-qt/sparkfun-qwiic) the quiic and stemmaqt are ordered the same. In fact they say:

> STEMMA QT is cross-compatible with Qwiic - STEMMA QT connector/cable is same as Qwiic. You can use STEMMA QT devices with Qwiic devices/controllers.

NightSkySK, 2021-03-05:
@daveismith There are two connector size promoted by Adafruit 
STEMMA (which is also compatible with Seeed Studio Grove ) Use JST PH 2.0mm pitch connectors
> Compatibility:
> 
> STEMMA 4-pin cables are cross-compatible with Grove parts. You can plug the larger 4-pin STEMMA cables/devices/controllers into Grove I2C devices/controllers just fine and vice-versa. Even though the connector they use is not the exact same part, the cables will fit into the connectors!
> STEMMA uses the same voltage as Grove - power is 3-5VDC and data is 3-5VDC with level shifting/regulators on devices.

STEMMA QT is cross-compatible with Qwiic use 4 Pin JST SH - 1.0mm pitch connectors
STEMMA QT due to its smaller size is more convenient to use on I2C sensors and devices, once STEMMA is commonly used on microcontroller boards.

daveismith, 2021-03-05:
They also suggest: https://www.sparkfun.com/products/15109 as an option to support Grove devices. The other thing to consider is that Grove isn't always I2C so you probably would need to create an adaptor to the aux header for anything else anyhow.

NightSkySK, 2021-03-05:
I've just found that there are Qwiic connectors on the schematic as J26 and J28 and SCL and SDA lines included also as part of AUX J25 and J27 at mobo-rev03 branch so it seems to be already implemented. For me, it doesn't matter to which size of the connector is going to be placed. More important for me was to not create any new standard but use something which is already well defined and commonly used in the market. I'm sorry if I created some confusion.

austinscreations, 2021-03-05:
I've only recently started following the index project, but saw some of the newer schematics on discord where IC's are being used on the I2C bus but having the address pins latched, while it's fine to do this my thought was if you leaving the option to breakout it could be worth having pads for those address pins should anyone run into conflicts later. (at the time of opening the ticket i hadn't actually looked to see if there were already breakouts on the MOBO - maybe I should have but it's not really the focus of the ticket)

daveismith, 2021-03-05:
@austinscreations - absolutely no issue with your request, it makes some sense though in discussion yesterday I suggested to @G-Pereira that we just put a jumper footprint (resistor with both pads connected by a trace which can be cut) in place to disconnect the vacuum sensor circuit as we're not 100% sure it's going to work. that way if it doesn't we can iterate externally until we get a workable solution and then bring it back on-board. I believe his plan was to make all of this circuitry a variant so that by default it wouldn't be included on builds until it's been validated.

G-Pereira, 2021-03-05:
[image]
Done!

closed 2021-03-05 by G-Pereira