opulo-inc/lumenpnp #212: [MOBO] - rev03 feedback 5V switcher - Vout selection resistors

Phase4logic, 2021-03-07:
This is my first feedback on the design.  Very nice work everyone.

I'm curious why 4 resistors (R30-R33) are used to set the voltage on the TPS54531 switcher?  I ran a couple of simulations on TI's WEBENCH Power Designer, using the suggested values of 10.2K and 1.96K from TI's datasheet, and Vout is almost exactly the same.  Both of these values are standard 1% values.

Using 2 resistors vs 4 would reduce the number of parts to and unique parts count by two.

wilhelmzeuschner, 2021-03-07:
This is done to reduce the amount of unique BOM parts/values. 10.2k and 1.96k would only be used for FB but would potentially introduce the need for two additional feeders just for the two resistors.

G-Pereira, 2021-03-07:
Thanks for your feedback @Phase4logic !
Exactly like @wilhelmzeuschner said: we had two resistors on this voltage divider but then in order to reduce our unique part count we replaced them with resistor values that we were already using elsewhere.

closed 2021-03-07 by G-Pereira