opulo-inc/lumenpnp #221: [CAD] - Add Z-Axis idler Spacer to Repo

daveismith, 2021-03-17:
A spacer for the z-axis idler doesn't exist in the cad but is something a number of builders have needed to add to their builds. One was posted in discord at [https://discord.com/channels/688723670461448225/739250421893300264/796448153565790248](https://discord.com/channels/688723670461448225/739250421893300264/796448153565790248). It should be added to the git repo and assembly.

vysocan, 2021-03-17:
Hello,
How nice. I was just collecting the z part. I've used the above cad file and created slightly modified one with nut pocket. It will allow the bolt to be firmly attached to the Y front gantry.
[image]
Also the pad is leaner to allow z gantry movement.
Actually my fist FreeCAD work :)
[Z-idler-standoff-v3.zip](https://github.com/sphawes/index/files/6159037/Z-idler-standoff-v3.zip)

G-Pereira, 2021-07-29:
Fixed!

PR #318 (2021-06-13): Belt Tensioning Update and Mechanical BOM Refresh
Hey all!

Here's a monster PR with all the recent CAD updates, along with an updated mechanical Bill of Materials. Here is what is included:

- Fixed improper method of building files in FreeCAD, now only basing sketches off of Datum Planes
- All belts have fine tension adjustment, either through a screw adjustment or a slotted holes for a stepper motor mount (Closes #229)
- All roller gantries have fine tensioning on both sides of the bolt holding the third roller
- All fasteners are included in the assembly file, generated by the Fastener wor [... cut]
closed 2021-07-29 by G-Pereira